FreeCAD/FreeCAD #13421: Material system: Material always openes the full tree (expanded)

M4xTheMan, 2024-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When setting a material to a body for the first time **or** when opening the material dialog for a body which already has a material assignment, the "material tree" always opens fully expanded.

**Proposal**

- If no material got assigned earlier (= default material is assigned): Expand only "Favorites" and "Recent"
- If a material is already assigned: Open the "category" where this material belongs to.

**Questions**

- How many materials would be displayed under "recent"?
- Can one material be present in several "categories" (apart from favorites and recent)? If so, it would be difficult to expand the corresponding category if a material is already assigned.
- If a material is already assigned: Would it be useful to "jump" to the assigned material?

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git) AppImage
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 11.1.0
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.2
  * Manipulator
  * fasteners 0.5.16
  * Assembly3 0.12.2
  * sheetmetal 0.4.10
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.29
  * POV-Ray-Rendering
  * OpenDark 2023.12.17
  * 3DfindIT 1.2.0
  * Gridfinity 0.3.0
  * Freecad-Built-in-themes-beta 1.2.2
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-04-13:
This was a known issue in the initial merge. It was intended as minimum viable product and is being addressed.

M4xTheMan, 2024-04-15:
Do you've the number of the PR at hand?

davesrocketshop, 2024-04-15:
> Do you've the number of the PR at hand?

It is in progress, not submitted yet. Expect it by tomorrow.

PR #13496 (2024-04-22): Material: MaterialTreeWidget usability enhancements
Improves the MaterialTreeWidget beyond minimum viable product.

- Filters can now be filter lists to allow a variety of filtering options.
- User preferences allow the inclusion/exclusion of favorites and recents.
- Widget state such as expansion, tree expansions, etc are saved and restored.
- show current appearancee material when editing.
- implements a python interface

Fixes #13421: always opens full tree
closed 2024-04-22 by chennes