FreeCAD/FreeCAD #13423: Docking: no way to resize `Report View` when `floating=true` and `overlay=true`

Andrei-Pozolotin, 2024-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PROBLEM:

Currently, when using an  `OverlayOption::Enable` in preferences:
[image]
then the standard docking windows such as:
* `Python Console`, `Report View`, `Model`, `Tasks` , etc.

after being switched into the `floating` (or detached) mode, 
will have  `Qt::FramelessWindowHint`  https://doc.qt.io/qt-5/qt.html#WindowType-enum set as well:

https://github.com/FreeCAD/FreeCAD/blob/9991c430d2b350ae672618d0f20c1527a607984e/src/Gui/OverlayWidgets.cpp#L1109
```cpp
        if(option != OverlayOption::Disable) {
            widget->setWindowFlags(widget->windowFlags() | Qt::FramelessWindowHint);
        } else {
            widget->setWindowFlags(widget->windowFlags() & ~Qt::FramelessWindowHint);
        }
```

with the result being that user not able:
*  to grab/drag the window by the border, 
* and hence nor to resize these windows any more.

SOLUTION:

Perhaps the "overlay" and the "floating" should be made mutually exclusive?
Perhaps the "overlay mode" should be emulated without using a `Qt::FramelessWindowHint`?

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36812 (Git)
Build type: None
Branch: makepkg
Hash: 9991c430d2b350ae672618d0f20c1527a607984e
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-27:
Is this still relevant ?

Andrei-Pozolotin, 2024-12-14:
yes, "zero border " makes it impossible to grab/resize the floating windows