FreeCAD/FreeCAD #13427: Errors when running with Qt 6.7

liamhays, 2024-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just upgraded my Arch Linux laptop to Qt 6.7 and FreeCAD seems to be having trouble with it. On startup, report view complains about PySide issues:

```
07:51:57  During initialization the error "'PySide6.QtWidgets.QStatusBar.removeWidget' called with wrong argument types:
  PySide6.QtWidgets.QStatusBar.removeWidget(QPushButton)
Supported signatures:
  PySide6.QtWidgets.QStatusBar.removeWidget(PySide6.QtWidgets.QWidget)" occurred in /usr/lib/freecad/Mod/Tux/InitGui.py
07:51:57  Please look into the log file for further information
```
In the log file is the full traceback:
```
Log: Traceback (most recent call last):
  File "<string>", line 136, in RunInitGuiPy
  File "<string>", line 26, in <module>
  File "/usr/lib/freecad/Mod/Tux/NavigationIndicatorGui.py", line 583, in <module>
    statusBar.removeWidget(label)
TypeError: 'PySide6.QtWidgets.QStatusBar.removeWidget' called with wrong argument types:
  PySide6.QtWidgets.QStatusBar.removeWidget(QPushButton)
Supported signatures:
  PySide6.QtWidgets.QStatusBar.removeWidget(PySide6.QtWidgets.QWidget)
Log: ----------------------------------------------------------------------------------------------------
Err: During initialization the error "'PySide6.QtWidgets.QStatusBar.removeWidget' called with wrong argument types:
  PySide6.QtWidgets.QStatusBar.removeWidget(QPushButton)
Supported signatures:
  PySide6.QtWidgets.QStatusBar.removeWidget(PySide6.QtWidgets.QWidget)" occurred in /usr/lib/freecad/Mod/Tux/InitGui.py
Err: Please look into the log file for further information
```

After showing the error, FreeCAD seems to mostly work (the clickable items on the Start page aren't clickable, but otherwise, it's functional). However, loading Assembly4 causes this error:
[image]

And trying to open the Addon Manager (because I thought maybe an addon update would fix the problem) causes this output in report view:
```
07:53:56  Running the Python command 'Std_AddonMgr' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/AddonManager/AddonManager.py", line 161, in Activated
    NetworkManager.InitializeNetworkManager()
  File "/usr/lib/freecad/Mod/AddonManager/NetworkManager.py", line 655, in InitializeNetworkManager
    AM_NETWORK_MANAGER = NetworkManager()
                         ^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/AddonManager/NetworkManager.py", line 171, in __init__
    self.QNAM.setCache(self.diskCache)

'PySide6.QtNetwork.QNetworkAccessManager.setCache' called with wrong argument types:
  PySide6.QtNetwork.QNetworkAccessManager.setCache(QNetworkDiskCache)
Supported signatures:
  PySide6.QtNetwork.QNetworkAccessManager.setCache(PySide6.QtNetwork.QAbstractNetworkCache)
```

It seems like something in FreeCAD isn't compatible with changes in PySide.

### Full version info

```shell
[code]
OS: Arch Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Pyramids-and-Polyhedrons
  * Assembly4 0.50.12
  * fasteners 0.5.17
[/code]
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-12:
Is this related to https://github.com/FreeCAD/FreeCAD/issues/13303 ? You are using the stable version, could you also try the weekly builds if they have the same error: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

AndrewAmmerlaan, 2024-04-13:
On Gentoo freecad fails to compile with qt/pyside 6.7.0:
```
FAILED: src/Gui/CMakeFiles/FreeCADGui.dir/PythonWrapper.cpp.o
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DBUILD_ADDONMGR -DCMAKE_BUILD_TYPE=\"RelWithDebInfo\" -DFMT_SHARED -DFreeCADGui_EXPORTS -DHAVE_CONFIG_H -DHAVE_FFMPEG -DHAVE_OPENGL_EXT -DHAVE_PYSIDE6 -DHAVE_RAPIDJSON -DHAVE_SHIBOKEN6 -DHAVE_TBB -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_OPENGLWIDGETS_LIB -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_SVGWIDGETS_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DSPNAV_FOUND -DVTK_OPENGL2_BACKEND -D_OCC64 -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/FreeCADGui_autogen/include -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build -I/usr/include/eigen3 -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui/Quarter -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui/.. -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/.. -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/Language -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/propertyeditor -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/TaskView -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/Quarter -I/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2_build/src/Gui/DAGView -I/usr/include/qt6/QtDBus -I/usr/include/qt6/QtUiPlugin -I/usr/include/PySide6/QtCore -I/usr/include/PySide6/QtGui -I/usr/include/PySide6/QtWidgets -isystem /usr/include/python3.11 -isystem /usr/include/qt6/QtCore -isystem /usr/include/qt6 -isystem /usr/include/qt6/QtWidgets -isystem /usr/include/qt6/QtGui -isystem /usr/include/qt6/QtOpenGL -isystem /usr/include/qt6/QtPrintSupport -isystem /usr/include/qt6/QtSvg -isystem /usr/include/qt6/QtSvgWidgets -isystem /usr/include/qt6/QtNetwork -isystem /usr/include/qt6/QtUiTools -isystem /usr/include/qt6/QtOpenGLWidgets -isystem /usr/include/shiboken6 -isystem /usr/include/PySide6 -isystem /usr/lib64/qt6/mkspecs/linux-g++-isystem /usr/include/qt6/QtXml  -Wall -Wextra -Wno-write-strings -O3 -pipe -march=skylake -mtune=skylake -std=gnu++17 -fPIC -fPIC -MD -MT src/Gui/CMakeFiles/FreeCADGui.dir/PythonWrapper.cpp.o -MF src/Gui/CMakeFiles/FreeCADGui.dir/PythonWrapper.cpp.o.d -o src/Gui/CMakeFiles/FreeCADGui.dir/PythonWrapper.cpp.o -c /tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui/PythonWrapper.cpp
/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui/PythonWrapper.cpp: In function ‘bool Gui::loadPySideModule(const std::string&, PyTypeObject**&)’:
/tmp/portage/media-gfx/freecad-0.21.2-r2/work/FreeCAD-0.21.2/src/Gui/PythonWrapper.cpp:481:43: error: cannot convert ‘Shiboken::Module::TypeInitStruct*’to ‘PyTypeObject**’ {aka ‘_typeobject**’} in assignment
481 |         types = Shiboken::Module::getTypes(requiredModule);
|                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
|                                           |
|                                           Shiboken::Module::TypeInitStruct*
```

liamhays, 2024-04-16:
An equivalent [Arch Linux package issue](https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/5) recommends downgrading to 0.21.2-6 until the issue is resolved. I can confirm that this works, but is obviously not a permanent solution.

luzpaz, 2024-04-17:
HI @adrianinsaval, are you still helping with the FC Arch package ?

liamhays, 2024-04-19:
pyside2 and friends are no longer orphaned in the AUR, but they will clearly stay there forever now.

I would also argue that the edited title ("issue blocking bleeding-edge builds") is not entirely correct: the Arch package that doesn't work is stable version 0.21.2. A bleeding-edge development branch build actually works fine (because it's actually built for Qt6), whether built from scratch or using the AppImage.

Personally, I'm ok with leaving `freecad` in the pacman `IgnorePkg` list until the next major release, so that it doesn't always try to upgrade to the half-broken 0.21.2-15 version. No, it's not ideal, but maybe that's the price of rolling releases.

YakoYakoYokuYoku, 2024-04-22:
I'm trying to tackle this issue and I've made some discoveries. The `TypeError`s we are seeing here are a because of a bug in Shiboken, I've used both `gdb` and `pdb` to verify that the return values of `Gui::PythonWrapper::fromQWidget` are correct and they seem to be so. Here's a sample `pdb` trace (though interleaved because of some other issues with the FreeCAD console) that shows the bug.

```
> /usr/lib/freecad/Mod/Tux/NavigationIndicatorGui.py(584)<module>()
-> print(label, isinstance(label, QtGui.QWidget))
p label
(Pdb) <PySide6.QtWidgets.QPushButton(0x57ebc20e2730) at 0x79e0825347c0>
import inspect
p type(label)
(Pdb) <class 'PySide6.QtWidgets.QPushButton'>
p type(statusBar)
(Pdb) <class 'PySide6.QtWidgets.QStatusBar'>
p inspect.getmro(type(label))
(Pdb) (<class 'PySide6.QtWidgets.QPushButton'>, <class 'PySide6.QtWidgets.QAbstractButton'>, <class 'PySide6.QtWidgets.QWidget'>, <class 'PySide6.QtCore.QObject'>, <class 'PySide6.QtGui.QPaintDevice'>, <class 'Shiboken.Object'>, <class 'object'>)
p inspect.getmro(type(statusBar))
(Pdb) (<class 'PySide6.QtWidgets.QStatusBar'>, <class 'PySide6.QtWidgets.QWidget'>, <class 'PySide6.QtCore.QObject'>, <class 'PySide6.QtGui.QPaintDevice'>, <class 'Shiboken.Object'>, <class 'object'>)
p isinstance(label, object)
(Pdb) True
p isinstance(label, QtGui.QWidget)
(Pdb) False
p inspect.getmro(QtGui.QWidget)
(Pdb) (<class 'PySide6.QtWidgets.QWidget'>, <class 'PySide6.QtCore.QObject'>, <class 'PySide6.QtGui.QPaintDevice'>, <class 'Shiboken.Object'>, <class 'object'>)
p isinstance(statusBar, QtGui.QWidget)
(Pdb) False
```

So this is not FreeCAD's fault but PySide6/Shiboken6 at version `6.7.0`.

YakoYakoYokuYoku, 2024-04-22:
I've got the upstream bug report ([PYSIDE-2674](https://bugreports.qt.io/browse/PYSIDE-2674)), Hurra!

liamhays, 2024-04-22:
Thank you for finding and reporting this! I assumed it was a FreeCAD issue or caused by premature updates to pyside, but apparently not.

YakoYakoYokuYoku, 2024-04-25:
Yikes! Opened #13647 for that.

liamhays, 2024-04-27:
The Arch package has been updated to 0.21.2-17, which backports some code and fixes the issues before: no errors on startup, the Addon Manager works, and so do plugins (and that's including the Arch upgrade to Python 3.12). I don't know if this issue should be closed yet, because it seems that the underlying issue is a whole can of worms, but I can confirm that FreeCAD is back to functional on Arch.

adrianinsaval, 2024-04-30:
update your system, partial upgrades are not supported in archlinux, you need the version compiled with python 3.12 and qt6

adrianinsaval, 2024-04-30:
the remaining issue here has been fixed by #13647, the upstream qt issue has been patched in archlinux and will be generally fixed with qt 6.7.1

closed 2024-04-30 by adrianinsaval

jock-tanner, 2024-05-04:
Hello, Arch user here, unfortunately 0.21.2-20 is still buggy (at least for me) after full system upgrade:

- nav cube is not working,
- `Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'` on application start,
- Dimension font is non-existent and therefore invisible, when trying to change it: `(qt.core.qobject.connect) QObject::connect: No such signal QComboBox::activated(const QString&)`,
- when opening old projects:
```
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Draft/draftviewproviders/view_base.py", line 179, in attach
    self.onChanged(vobj, "Pattern")
  File "/usr/lib/freecad/Mod/Draft/draftviewproviders/view_base.py", line 286, in onChanged
    if str(vobj.Pattern) in list(utils.svg_patterns().keys()):
                                 ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 789, in svg_patterns
    loadSvgPatterns()
  File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 763, in load_svg_patterns
    p = importSVG.getContents(file, 'pattern')
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/Draft/importSVG.py", line 1709, in getContents
    f = pythonopen(filename)
        ^^^^^^^^^^
<class 'NameError'>: name 'pythonopen' is not defined
```
- a lot of `SyntaxWarning: invalid escape sequence` but that's probably expected

I'm not sure which of that make sense as a separate bug report. I was a Python dev once, but now I'm a carpenter. Please help me sort this out. I'm building a house, and all my plans are in FreeCAD.

YakoYakoYokuYoku, 2024-05-04:
@jock-tanner, I'm looking for a fix to your problems, though, it'd be a better idea to open an issue at Arch's Gitlab [freecad repository](https://gitlab.archlinux.org/archlinux/packaging/packages/freecad).

adrianinsaval, 2024-05-06:
note that we have not claimed that archlinux's 0.21 package is fixed. FreeCAD main branch has one of the issues described above fixed and the other has been fixed upstream in qt6.

Please do not mix other archlinux qt6 related bugs here, at this point it just makes things more confusing. #13303 is more appropriate. Either way I will ask for a little patience as I will be making a new release in the 0.21 series soon with some of the recent fixes backported