FreeCAD/FreeCAD #13428: [Feature Request] Add Projection tool to Part Design

NXTler, 2024-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Greetings,
As of now, the only way to do projections in Part Design is through the "Create Projection on Surface" tool from Part (essentially using both workbenches). This makes modeling unnecessary complex and finicky in Part Design. So in an effort to only have one Workbench in the future, it could be a good first step to port this feature over to Part Design.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git) AppImage
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.4.2
  * A2plus 0.4.63
  * QuickMeasure 2022.10.28
  * Curves 0.6.23
  * CfdOF 1.25.2
  * IconThemes
  * fasteners 0.5.12
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-12:
What do you want to project? Images, sketches, selected geometry, features from a specific angle? Or just use the Part tool inside Part Design as a  parametric feature?

NXTler, 2024-04-12:
Sorry for not being specific enough, the original tool can project edges, wires and faces. I'm mostly interested in wires (for sketches), but it would be nice to also have the other options.