FreeCAD/FreeCAD #13431: Referenced images cannot be updated when changed externally

maxwxyz, 2024-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Images that get updated on disk cannot be updated in FreeCAD**
External images from the local machine which are used as reference images with a relative file path won't update in FreeCAD if they get modified externally by other software. There is no possibility to update them inside FreeCAD. Currently they are contained in the FCStd file but they also contain a reference so then they could auto reload.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * OpticsWorkbench 1.0.15
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-18:
Related:

* https://github.com/FreeCAD/FreeCAD/issues/5746

luzpaz, 2024-09-26:
Hey @mnesarco, you have some familiarity with this approach due to the ability of guitar wb to alter inkscape data within FreeCAD (live but with latency) correct?  Would you be open to write out how to technically this could be made to happen ?

mnesarco, 2024-09-26:
Hello @luzpaz ,
Are you talking about FreeCAD -> Inkscape or Inkscape -> FreeCAD or both?

To detect file changes I have used `QFileSystemWatcher` it works.

luzpaz, 2024-09-30:
@mnesarco both. Would you mind linking to the code? (for posterity)

mnesarco, 2024-09-30:
> @mnesarco both. Would you mind linking to the code? (for posterity)

That code is not used right now in Marz, in the current version the svg file is not monitored for changes. But the code is there anyway (old deprecated code):  https://github.com/mnesarco/MarzWorkbench/blob/master/freecad/marz/utils/filesystem.py

Hot reload from any external file (ie inkscape saved svg) to FreeCAD is possible using QFileSystemWatcher.
But FreeCAD to Inkscape is not possible due to inkscape limitations. Inkscape does not run python in the same way as FreeCAD does.

The QFileSystemWatcher api is quite simple anyway:

```python

def on_file_changed(path):
  print(f"{path} Changed")

WATCHER = QtCore.QFileSystemWatcher(Gui.getMainWindow())
WATCHER.fileChanged.connect(on_file_changed)
WATCHER.addPath('/path/to/my/file/or/dir')

```