FreeCAD/FreeCAD #13435: Material list editor issues

davesrocketshop, 2024-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a known issue that existed when the original code was checked in but which does not have a corresponding issue tag.

The list editor covers all types of lists: string, file, and image.

There are a number of issues with the editor. Adding a row, particularly with file lists, is difficult. Selecting a new row is not enough to add a list. Conversely, in the case of strings, extra empty rows are often added at the end of the list. Image lists have no corresponding editor.

Problems are compounded when deleting entries. The delete key will often delete the entry without deleting the row. There should be a separate button to perform that operation.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36757 +32 (Git)
Build type: Unknown
Branch: issue13019_20240406
Hash: d1391272e64fe2f1c0253d0d1f99a5201e4dda68
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * Rocket 3.2.4
  * Render 2023.8.18
  * parts_library
  * parts_library.backup1693954863.8543406 (Disabled)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-06:
I've added check-boxes to this ticket in order to track the meta-issues mentioned in it.

luzpaz, 2026-01-01:
@davesrocketshop you got this or can we label it 'good first issue' to signal folks who may be open to helping realize this effort ?