FreeCAD/FreeCAD #13446: Crash with hole in additive loft

Lykos153, 2024-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to make a hole into an additive loft when FreeCAD crashes on me.

Steps to reproduce:
1. Open file
2. Go to PartDesign
3. Select Additive Loft
4. Select Hole tool
5. Crash

```
Program received signal SIGSEGV, Segmentation fault.
#0  /nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6(+0x3ff30) [0x7ff22daa8f30]
#1  0x7ff1c8d627d8 in PartDesign::Hole::findHoles(TopoDS_Shape const&, TopoDS_Shape const&) const from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/_PartDesign.so+0x308
#2  0x7ff1c8d65dfb in PartDesign::Hole::execute() from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/_PartDesign.so+0xcbb
#3  0x7ff2312fbeea in App::DocumentObject::recompute() from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADApp.so+0xba
#4  0x7ff188b608bb in Part::Feature::recompute() from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/Part.so+0xb
#5  0x7ff23128bffc in App::Document::_recomputeFeature(App::DocumentObject*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADApp.so+0x2ec
#6  0x7ff231290b5a in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADApp.so+0x5ba
#7  0x7ff231367a4c in App::DocumentPy::recompute(_object*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADApp.so+0x2ec
#8  0x7ff231367b63 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADApp.so+0x23
#9  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(+0x24ab02) [0x7ff23004ab02]
#10  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7ff2300016bd]
#11  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x33cc) [0x7ff22fef7c4c]
#12  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(+0x32c7ec) [0x7ff23012c7ec]
#13  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(PyEval_EvalCode+0x14a) [0x7ff23012c98a]
#14  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(+0x3576a0) [0x7ff2301576a0]
#15  /nix/store/gd3shnza1i50zn8zs04fa729ribr88m9-python3-3.11.8/lib/libpython3.11.so.1.0(PyRun_StringFlags+0x73) [0x7ff230157d33]
#16  0x7ff230be74a8 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADBase.so+0x68
#17  0x7ff231e97869 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x89
#18  0x7ff231e979f0 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0xe0
#19  0x7ff231e97bd2 in Gui::Command::updateActive() from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x42
#20  0x7ff17c528875 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/PartDesignGui.so+0x2d5
#21  /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/PartDesignGui.so(+0x130557) [0x7ff17c530557]
#22  /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/PartDesignGui.so(+0x12dbcd) [0x7ff17c52dbcd]
#23  0x7ff17c535475 in prepareProfileBased(PartDesign::Body*, Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (Part::Feature*, App::DocumentObject*)>) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/PartDesignGui.so+0x365
#24  0x7ff17c536455 in CmdPartDesignHole::activated(int) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/PartDesignGui.so+0xe5
#25  0x7ff231e9efb6 in Gui::Command::_invoke(int, bool) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x226
#26  0x7ff231e9f35e in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x12e
#27  /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Core.so.5(+0x3145d2) [0x7ff22e1145d2]
#28  0x7ff22ed6a502 in QAction::triggered(bool) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x42
#29  0x7ff22ed6d05f in QAction::activate(QAction::ActionEvent) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0xaf
#30  /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5(+0x26444a) [0x7ff22ee6444a]
#31  0x7ff22ee64597 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0xf7
#32  0x7ff22ef5d34a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0xa
#33  0x7ff22edb29a8 in QWidget::event(QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x268
#34  0x7ff22ed70dce in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x7e
#35  0x7ff22ed7913f in QApplication::notify(QObject*, QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x10ef
#36  0x7ff231e47ab8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x98
#37  0x7ff22e0dbc58 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Core.so.5+0x118
#38  0x7ff22ed7750e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x1be
#39  /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5(+0x1cb44d) [0x7ff22edcb44d]
#40  /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5(+0x1ce79f) [0x7ff22edce79f]
#41  0x7ff22ed70dce in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Widgets.so.5+0x7e
#42  0x7ff231e47ab8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /nix/store/mbk3i1fgbyinmiv4rz89p15dl3qcqy5h-freecad-0.21.2/lib/libFreeCADGui.so+0x98
#43  0x7ff22e0dbc58 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Core.so.5+0x118
#44  0x7ff22e52fa3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Gui.so.5+0x87b
#45  0x7ff22e502cdc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Gui.so.5+0xac
#46  /nix/store/y3qcljhcp4maxw99qk83ncf22rpbmfg1-qtbase-5.15.12-bin/lib/qt-5.15.12/plugins/platforms/../../../../../y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5XcbQpa.so.5(+0x6b0aa) [0x7ff21ac050aa]
#47  /nix/store/q6jyzkl2f1capj5yc0rq65i0zfm9b82k-glib-2.78.4/lib/libglib-2.0.so.0(+0x5a744) [0x7ff22a312744]
#48  /nix/store/q6jyzkl2f1capj5yc0rq65i0zfm9b82k-glib-2.78.4/lib/libglib-2.0.so.0(+0x5d797) [0x7ff22a315797]
#49  /nix/store/q6jyzkl2f1capj5yc0rq65i0zfm9b82k-glib-2.78.4/lib/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7ff22a315d8c]
#50  0x7ff22e1351b6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Core.so.5+0x66
#51  0x7ff22e0da67b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/y3w2wzj83525frlak51nmjsvj2xialw0-qtbase-5.15.12/lib/libQt5Core.so.5+0x13b
[... cut ...]
#55  /nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6(+0x2a10e) [0x7ff22da9310e]
#56  /nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6(__libc_start_main+0x89) [0x7ff22da931c9]
#57  freecad() [0x408fb5]
```

_May_ be connected to #13107 

### Full version info

```shell
`FreeCAD 0.21.2 Revision: Unknown` on NixOS
```

### Subproject(s) affected?

PartDesign

### Anything else?

[additiveloft.zip](https://github.com/FreeCAD/FreeCAD/files/14968956/additiveloft.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-13:
Full About info please.

Lykos153, 2024-04-13:
Added

maxwxyz, 2024-04-14:
Could you try this with the current weekly builds?

xtemp09, 2024-04-14:
The crash occurs in function `PartDesign::Hole::findHoles`, refactored by ven. wwmayer in the commit b05fb731962b73b63204992d4a92f39a91b5da9b from the code written by Eivind Kvedalen in the commit 04b176ebfb9c6a372c7f48074eeb687618dff315.

<details> <summary>Backtrace</summary>

```cpp
Process 1732 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff6d0cb229 _PartDesign.so`PartDesign::Hole::findHoles(this=0x000055555a18ad30, profileshape=0x00007fffffff9650, protohole=0x00007fffffff9770) const at FeatureHole.cpp:2003:27
   2000         Handle(Geom_Curve) c = BRep_Tool::Curve(edge, c_start, c_end);
   2001
   2002         // Circle?
-> 2003         if (c->DynamicType() != STANDARD_TYPE(Geom_Circle)) {
   2004             continue;
   2005         }
   2006
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff6d0cb229 _PartDesign.so`PartDesign::Hole::findHoles(this=0x000055555a18ad30, profileshape=0x00007fffffff9650, protohole=0x00007fffffff9770) const at FeatureHole.cpp:2003:27
    frame #1: 0x00007fff6d0c9a59 _PartDesign.so`PartDesign::Hole::execute(this=0x000055555a18ad30) at FeatureHole.cpp:1886:66
    frame #2: 0x00007ffff4bd2c3c libFreeCADApp.so`App::DocumentObject::recompute(this=0x000055555a18ad30) at DocumentObject.cpp:161:29
    frame #3: 0x00007fff6d023c32 _PartDesign.so`PartDesign::Feature::recompute(this=0x000055555a18ad30) at Feature.cpp:80:37
    frame #4: 0x00007ffff4ae0067 libFreeCADApp.so`App::Document::_recomputeFeature(this=0x0000555558d5f650, Feat=0x000055555a18ad30) at Document.cpp:3129:41
    frame #5: 0x00007ffff4add77e libFreeCADApp.so`App::Document::recompute(this=0x0000555558d5f650, objs=size=0, force=false, hasError=0x0000000000000000, options=0) at Document.cpp:2856:48
    frame #6: 0x00007ffff4ca505a libFreeCADApp.so`App::DocumentPy::recompute(this=0x0000555558c8fc20, args=0x00007fffecd9c070) at DocumentPyImp.cpp:584:54
    frame #7: 0x00007ffff4c9bde3 libFreeCADApp.so`App::DocumentPy::staticCallback_recompute(self=0x0000555558c8fc28, args=0x00007fffecd9c070) at DocumentPy.cpp:1952:66
    frame #8: 0x00007ffff3528008 libpython3.10.so.1.0`___lldb_unnamed_symbol3713 + 72
    frame #9: 0x00007ffff34dfadc libpython3.10.so.1.0`_PyObject_MakeTpCall + 140
    frame #10: 0x00007ffff347ba1c libpython3.10.so.1.0`_PyEval_EvalFrameDefault + 40380
    frame #11: 0x00007ffff35c23af libpython3.10.so.1.0`___lldb_unnamed_symbol4418 + 1647
    frame #12: 0x00007ffff35bd3de libpython3.10.so.1.0`PyEval_EvalCode + 190
    frame #13: 0x00007ffff360c4cd libpython3.10.so.1.0`___lldb_unnamed_symbol4668 + 157
    frame #14: 0x00007ffff360d4f9 libpython3.10.so.1.0`PyRun_StringFlags + 121
    frame #15: 0x00007ffff3d01ef6 libFreeCADBase.so`Base::InterpreterSingleton::runString[abi:cxx11](this=0x000055555562b750, sCmd="App.ActiveDocument.recompute()") at Interpreter.cpp:249:15
    frame #16: 0x00007ffff67baf4b libFreeCADGui.so`Gui::Command::_runCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Command.cpp", line=851, eType=App, sCmd="App.ActiveDocument.recompute()") at Command.cpp:725:38
    frame #17: 0x00007ffff67bab73 libFreeCADGui.so`Gui::Command::_doCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Command.cpp", line=851, eType=App, sCmd="App.ActiveDocument.recompute()") at Command.cpp:677:16
    frame #18: 0x00007ffff67bbf3a libFreeCADGui.so`Gui::Command::updateActive() at Command.cpp:851:5
    frame #19: 0x00007fff56d6cd91 PartDesignGui.so`finishFeature(cmd=0x0000555559c7db20, Feat=0x000055555a18ad30, prevSolidFeature=0x0000000000000000, hidePrevSolid=true, updateDocument=true) at Command.cpp:548:26
    frame #20: 0x00007fff56d70432 PartDesignGui.so`finishProfileBased(cmd=0x0000555559c7db20, sketch=0x0000555559d445d0, Feat=0x000055555a18ad30) at Command.cpp:949:18
    frame #21: 0x00007fff56d70d65 PartDesignGui.so`operator(__closure=0x00007fffffffb8b0, sketch=0x0000555559d445d0, Feat=0x000055555a18ad30) at Command.cpp:1081:27
    frame #22: 0x00007fff56d803f8 PartDesignGui.so`std::__invoke_impl<void, CmdPartDesignHole::activated(int)::<lambda(Part::Feature*, App::DocumentObject*)>&, Part::Feature*, App::DocumentObject*>((null)=__invoke_other @ 0x00007fffffffb0d0, __f=0x00007fffffffb8b0, (null)=0x00007fffffffb160, (null)=0x00007fffffffb158) &) at invoke.h:61:36
    frame #23: 0x00007fff56d7e7d7 PartDesignGui.so`std::__invoke_r<void, CmdPartDesignHole::activated(int)::<lambda(Part::Feature*, App::DocumentObject*)>&, Part::Feature*, App::DocumentObject*>(__fn=0x00007fffffffb8b0, (null)=0x00007fffffffb160, (null)=0x00007fffffffb158) &) at invoke.h:111:28
    frame #24: 0x00007fff56d7ced3 PartDesignGui.so`std::_Function_handler<void(Part::Feature*, App::DocumentObject*), CmdPartDesignHole::activated(int)::<lambda(Part::Feature*, App::DocumentObject*)> >::_M_invoke(__functor=0x00007fffffffb8b0, __args#0=0x00007fffffffb160, __args#1=0x00007fffffffb158) at std_function.h:290:30
    frame #25: 0x00007fff56d886fd PartDesignGui.so`std::function<void (Part::Feature*, App::DocumentObject*)>::operator()(this=0x00007fffffffb8b0, __args#0=0x0000555559d445d0, __args#1=0x000055555a18ad30) const at std_function.h:591:9
    frame #26: 0x00007fff56d6ecc9 PartDesignGui.so`operator(__closure=0x00007fffffffb880, feature=0x0000555559d445d0, subs=size=0) at Command.cpp:781:13
[... cut ...] fffffdd88) at libc-start.c:392:3
    frame #81: 0x000055555557eca5 FreeCAD`_start + 37
(lldb) fr v c
(opencascade::handle<Geom_Curve>) c = (entity = 0x0000000000000000)
(lldb)
```

</details>

<details>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Debug
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-04-14:
@wwmayer fyi

commit c7c4f0d7 (2024-04-14): PD: Fixes #13446: Crash with hole in additive loft
commit e05e17c9 (2024-04-14): PD: Fixes #13446: Crash with hole in additive loft
closed 2024-04-14 by wwmayer