FreeCAD/FreeCAD #13448: TechDraw: LengthDimension: argument not found

johnsonm, 2024-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In https://gitlab.com/mcdanlj/MinimalBroach I have a TechDraw drawing in which three LengthDimensions are correctly displayed, but are marked as errors in red, and the console has warnings:

```
21:16:36  575.06 <App> Document.cpp(3152): exception in Unnamed1#Dimension003 thrown: argument not found
21:16:36  575.06 <App> Document.cpp(3152): exception in Unnamed1#Dimension004 thrown: argument not found
21:16:36  575.06 <App> Document.cpp(3152): exception in Unnamed1#Dimension005 thrown: argument not found
21:16:36  Dimension003: argument not found
21:16:36  Dimension004: argument not found
21:16:36  Dimension005: argument not found
21:17:34  C++ exception thrown (argument not found)
```

In addition, attempting to export to PDF from the drawing creates only a blank page.

Here's the version in which I see the problem:

https://gitlab.com/mcdanlj/MinimalBroach/-/raw/037f3de3699de915681950bdd4fee37341ee7a6e/MinimalBroach.FCStd

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git) AppImage
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Assembly3 0.12.2
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-04-14:
That document was done from beginning to end in a single session of FreeCAD.

Restarting the same version of FreeCAD, deleting those dimensions, and recreating them seems to have worked around the problem.

Hopefully the link to the old version of the file shows what went wrong. :crossed_fingers:

johnsonm, 2024-04-15:
I have now updated to the latest weekly, as well as simplified this part, and this error is applying all dimensions, including horizontal and vertical dimensions.

A version displaying this problem is available here:

https://gitlab.com/mcdanlj/MinimalBroach/-/blob/e31d8dddcaa941e13008e0354d60eb1538f3ada2/MinimalBroach.FCStd

It was saved from:

OS: Fedora Linux 39 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git) AppImage
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Assembly3 0.12.2
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)

WandererFan, 2024-04-16:
The dimensions with the errors all have references with high indices - Edge137, Vertex114, etc.

There are only 6 vertices and 6 Edges in Front.   Were the dimensions originally created in a more complex view with more geometric elements?  

Even in the 3d view, Holder only has 12 edges.

Would love to know how the dimensions were created with such odd references.

johnsonm, 2024-04-16:
I meant the bug report to be specifically about the `argument not found` exceptions from the initial version, and I think you are looking at the later version that I provided in case it was useful for diagnosis. Sorry for any confusion there. ☹

The first raw download link 037f3 has parts with modeled threads, which I expect are the source of the high index counts.

[image]

The second version has that hole removed from the model. It sounds like the problems in the second version e31d8, in the later comment, are just TNP, and it wasn't obvious to me from how they were displayed.

For that **initial** version with the modeled threads, I believe that I added and removed views a few times before I got the set I wanted. But that was an older weekly, so I could imagine that the `argument not found` is a bug since fixed...

If no one else has a similar bug, this is certainly reasonable to close as "could not reproduce" and you can just have the record here in case a pattern shows up later.

WandererFan, 2024-04-17:
No, no, it was a real bug.  Keep reporting them!  :)

The code that produces the error message when a geometry index is greater than the number of geometry entries had a subscript error.   Should be fixed by PR#13514.

johnsonm, 2024-04-18:
Thank you! You keep making TechDraw awesome.

I'll test it with the next weekly build and close this out.

luzpaz, 2024-04-19:
@johnsonm Do you mind if we close this and if you can recreate please re-open ?

closed 2024-04-19 by luzpaz

johnsonm, 2024-04-19:
Not in the least, I assumed it was left open because of wanting me to test first, otherwise I would have closed it proactively. ☺

johnsonm, 2024-04-25:
Just for completeness, I confirm this is fixed. Thanks again!