FreeCAD/FreeCAD #13457: PyInstaller can't add FreeCAD modules to EXE file

PeterBan11, 2024-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Firstly, my code was:

import sys
cwd = os.getcwd()
libPath = cwd+"\\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\\lib"
binPath = cwd+"\\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\\bin"
if os.path.isdir(libPath):
    sys.path.append(libPath)
if os.path.isdir(binPath):
    sys.path.append(binPath)
import FreeCAD
import Mesh
import Part

I thought that problem occurs because I append path to FreeCad in code.
So I have added in System PATH FreeCAD's path:
[image]

After that i tried this import code instead:

import FreeCAD
import Mesh
import Part

But it says that python can't locate module FreeCAD:
[image]

### Full version info

```shell
FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-14:
Please consult the forum for help too regarding this

lyphrowny, 2024-04-16:
I couldn't get it to work with conda build either, but it did work with a self-built version of FreeCAD: adding paths to `sys.path` was sufficent

maxwxyz, 2024-04-16:
so could we close this issue?

luzpaz, 2024-07-12:
@lyphrowny can you specifically show what the solution you arrived to? (for posterity)

lyphrowny, 2024-07-14:
@luzpaz 
Was only going to write the solution for the self-built FreeCAD, but decided to test it out with the weekly build and it worked. Don't know why it didn't work back then.
The only difference between the solutions for the buids is getting the builds :-) The core idea is the same: add binaries to the path

<details><summary>Self-built FreeCAD</summary>
<p>

1. Download the git repo of FreeCAD (let's say it was downloaded to `E:\FreeCAD` path)
2. Build FreeCAD (I built mine under `E:\FreeCAD\build")
3. Use the Python version, used by FreeCAD<sup>[1](#sbnote1)</sup>
4. Add a path with binaries to `sys.path` and try importing (should work :-)
```python
import sys

sys.path.append(r"E:\FreeCAD\build\bin")

import FreeCAD
import Part
import Mesh
```
___

<a name="sbnote1">1</a>: If you're using a different version, you'll get an error, telling you which version to use instead (Python3.8 in my case): 
```
ImportError: Module use of python38.dll conflicts with this version of Python.
```

</p>
</details> 

<details><summary>Weekly build of FreeCAD</summary>
<p>

1. Download a weekly build (for instance, [from here](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds))
2. Unpack it (let's say you unpacked it into "E:\Downloads" and the resulting path is "E:\Downloads\FreeCAD_weekly-builds-38127-conda-Windows-x86_64-py311")
3. Use the Python version, used by FreeCAD<sup>[1](#wbnote1)</sup>
4. Add a path with binaries to `sys.path` and try importing (should work :-)
```python
import sys

sys.path.append(r"E:\Downloads\FreeCAD_weekly-builds-38127-conda-Windows-x86_64-py311\bin")

import FreeCAD
import Part
import Mesh
```
___

<a name="wbnote1">1</a>: If you're using a different version, you'll get an error, telling you which version to use instead (Python3.11 in my case): 
```
ImportError: Module use of python311.dll conflicts with this version of Python.
```

</p>
</details>

closed 2024-07-15 by maxwxyz