FreeCAD/FreeCAD #13460: No self contained build system to set up simple build environment

maxwxyz, 2024-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**No self contained build system to set up simple build environment**
Discussion:
For example, there's no easy way of just downloading, e.g. VS Code setting or some simpler pre configured build system for click and run. Obviously it doesn't build. New contributors don't know how to fix it. It would be great to have a self contained build system; just import then run and make sure that the OS does not matter. That would be beneficial that more users can just test new changes and PRs.
So we have no self-contained build system that somebody can just pick up and build things without setting up a huge change.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * OpticsWorkbench 1.0.15
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

80avin, 2024-04-15:
+1 for this.
I have been reading the PRs in this repo for a long and tried to contribute twice mostly in Addon Manager, but even after wasting 2 days of my weekend, I couldn't get it to work locally. Went through forum and wikis to see all docker/devcontainer or any instructions, but either they were outdated, or not complete enough for me to catch.

I would really appreciate something on the lines of

1. [devcontainer](https://www.youtube.com/watch?app=desktop&v=5EjjTQAa0Nc) to build & run the entire codebase
2. something to make it easier to contribute to smaller parts like addon manager, etc
   1. Not sure what would be the ideal solution. A the minimum, maybe guides/documentation/video to demonstrate what would be ideal setup if someone wants to contribute in a smaller section.

dustinlacewell, 2024-04-16:
Hey, I recently rewrote the instructions for compiling on Windows to tackle the common path without getting lost in all the possible unique cases and options.

https://gist.github.com/dustinlacewell/708b5bc616b583e2e03437b31844b638

I've run through it a number of times and it seems to work pretty good. If you give it a shot, let me know!

glechapelain, 2024-07-05:
> Hey, I recently rewrote the instructions for compiling on Windows to tackle the common path without getting lost in all the possible unique cases and options. [...]

This is so interesting!  I was going to create a new topic in the forum about my situation and came across you guys' work.

So for me: Everything has been hunky-dory from the instructions from the Wiki up until those points:

1. I cannot debug Python code from what I have debugged.  I tried putting breakpoints both in the python source files in my git local copy and the one that gets copied over the build directory
2. I can run the Debug configuration, but not the release (some error/unhandled exception about mutex.)  Which I was thinking would indeed have to do with the LibPack dependencies compiled against some other configuration (debug instead of release.  But then it would make more sense that it was defaulted in LibPack to release, as @dustinlacewell instructions indicate)

I *am* able to debug in the Python source files from the official FreeCAD package °_°

hyarion, 2025-06-10:
@maxwxyz can't this be closed now as we have pixi?

closed 2025-06-10 by maxwxyz