FreeCAD/FreeCAD #13462: Long Python Script appears to lock-up FreeCAD

maxwxyz, 2024-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**When I have a long running python script FreeCAD wants to force quit, FreeCAD believes it has locked up**
Discussion:
When I have a long running Python script, FreeCAD keeps wanting me to close it, forcing quit and also there's no progress when it's printing stuff out I'm not seeing a progress of it printing the stuff out, it just stalls completely.
So when running a long running Python script, it appears to lock up, FreeCAD believes it has locked up and it does not tell you what it's actually doing.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * OpticsWorkbench 1.0.15
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-14:
Maybe one can provide a script for confirmation

luzpaz, 2024-08-14:
@yorikvanhavre have you seen this bug before ?