FreeCAD/FreeCAD #13467: [UI] Vertex shows as selected whenever hovering above any geometry, after losing mouse click effectiveness with settings opened

YosypC, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/50027752/668ccb60-4a1d-4540-8998-54ef6d1eba66

After opening the FC settings, my 21.3 Linux Mint behaved abnormally: mouse clicks did nothing in any part of the screen, Windows + D only temporarily showed the desktop, only to return to the settings page. I had to press Esc to close the settings window to regain control, it could be related since right after, this problem arose. Unfortunately I didn't think of using my phone to record, nor do I find a way to reproduce. I specifically used the "Display" -> "Light sources" page.

The Vertex7 (coincident to the origin, placed by the centered rectangle) always shows as selected even though it doesn't behave as such. Click anywhere yields no results, the only solution is to make a selection box. Hovering any geometry and clicking a void results in the vertex being "selected" again. Deleting the vertex fixes everything.

[ExtruderMount (copy).zip](https://github.com/FreeCAD/FreeCAD/files/14974901/ExtruderMount.copy.zip) Rename format

Sorry for the messy title.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36789 (Git)
Build type: Unknown
Branch: main
Hash: ff2bca2505010f580c9c35abfdfe8507e1c0d1cb
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-15:
@PaddleStroke FYI

luzpaz, 2024-07-29:
jfyi @PaddleStroke

closed 2025-09-08 by github-actions[bot]