FreeCAD/FreeCAD #13468: Sketcher: Finding specific constraint is difficult

maxwxyz, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Finding specific constraint to remove it is very hard**

Discussion:
With lots of constraints in a sketch and I then want to adjust the sketch and don’t want this constraint, it takes me about half an hour to find the constraint that I don’t want anymore.
When you set a bunch of constraints and you want to remove a specific one, then it's very hard to identify it and remove it.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * fasteners 0.5.16
  * OpenDark 2023.12.17
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

M4xTheMan, 2024-04-15:
Is [Sketcher_SelectConstraints](https://wiki.freecad.org/Sketcher_SelectConstraints) helpful in this case?

pierreporte, 2024-08-23:
`Sketcher_SelectConstraints` definitely helps, but once all constrains are selected they can still be hard to browse in the panel. There could be a way to filter the list to relevant constrains. Making that list filtered by default would make it more useful: its contents would change dynamically with the selection, and it would be the full list when nothing is selected.

luzpaz, 2024-10-20:
Is  there a proposal someone wants to put forward?