FreeCAD/FreeCAD #13469: Sketcher: Redundant / over constaint hard to identify

maxwxyz, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Constraint solver doesn't make it obvious where the over constraint is**

Discussion:
Auto remove redundant constraints is not enabled by default.
You can still become overconstrained if you have a constraint that sets more than one dimension.
(Overlapping explanations here about how overconstraints happen, sometimes the solver misreports overconstrained, etc.)
The constraint solver doesn't make it obvious what you've overconstrained when you've overconstrained.
And even when it does, it says “you need to delete this constraint” but I might want that one.
In the task dialog the redundant constraints are shown and a click selects them all but it is not easy identifyable in the 3D view.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * fasteners 0.5.16
  * OpenDark 2023.12.17
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Sketcher

### Anything else?

Related:
- https://github.com/FreeCAD/FreeCAD/issues/13468

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-21:
When tackling this issue, also consider:
- https://github.com/FreeCAD/FreeCAD/issues/13557

pierreporte, 2024-04-21:
It would be interesting if only over-constrained geometry was highlighted by color. Now, the whole sketch turns orange. Catia works this way: over-constrained geometry and redundant constrains turn magenta while the rest of the sketch don’t change. Though this needs #12036 and to avoid completely to rely on colors for anything but constrained status.

Automatic deletion of redundant constrains is not necessarily the best feature. The user may be more willing to delete the newly added constrain rather than losing the old ones.