FreeCAD/FreeCAD #13470: Points: Improvements for point clouds for reverse engineering

maxwxyz, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Reverse engineering when working point cloud is poor, no tools for exporting measurement from points**

Discussion:
There is no tool for extracting reference planes and measurements of points for reverse engineering when working with point clouds.
There is also no way to define a plane from the points.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git)
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * fasteners 0.5.16
  * OpenDark 2023.12.17
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Related:
- https://github.com/FreeCAD/FreeCAD/issues/13438

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pitch-circle, 2024-04-19:
> There is also no way to define a plane from the points.

The documentation for this macro suggests that statement may not be correct:

https://wiki.freecad.org/index.php?title=Macro_WorkFeatures

See also:

https://forum.freecad.org/viewtopic.php?t=11165

maxwxyz, 2024-04-19:
That is an 3rd party addon

luzpaz, 2024-10-16:
@mwganson any interest in this ?

mwganson, 2024-10-16:
In Part workbench menu there is a tool to create a points object from a mesh.  This gives selectable points that can be used to aid in remodeling a mesh, making measurements between points, etc.  This tool also works on Points objects from Points workbench.  The object produced is not parametric,meaning if the original mesh or points cloud changes, the object will not change with it, but rather would need to be recreated.