FreeCAD/FreeCAD #13471: TechDraw Line Width doesn't work for Views

ajohn77, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

No change visible after changing the "Line Width" of the View in TechDraw.
The dimensions "line width" work as expected on the other hand.

When exported to PDF, the line width seems very thin too, rendering it unusable for practical purposes.

Also the "Line Width Group" set in Preferences [TD->Annotation] makes no change, the new view is created with proper Line Width (according to properties->Data->LineWith), but it still displays inappropriate thin line.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

example project : [line_width_bug.zip](https://github.com/FreeCAD/FreeCAD/files/14976519/line_width_bug.zip)

[image]

exported PDF:  [line_width.pdf](https://github.com/FreeCAD/FreeCAD/files/14976580/line_width.pdf)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-04-15:
Changing the Line Group in preferences only applies the default to new views.  Once the view has been created, the values in the the View tab are the ones to change. 

Changing the values works here with:

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * BOLTSFC 2022.11.5
  * sheetmetal 0.2.63
  * fasteners 0.4.56
  * FreeCAD_Electric
  * freecad.gears 1.0.0
  * Curves 0.6.29
  * Assembly4 0.50.9
  * A2plus 0.4.64

ajohn77, 2024-04-17:
**It seems this is obviously issue of PPA version.**

I can confirm that "the same" 0.21.2 version works when downloaded as AppImage, but from PPA it doesn't:

**Not working for me:**
[code]
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
**Version: 0.21.2.33771 (Git)**
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

**Works normally:**
[code]
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
**Version: 0.21.2.33771 (Git) AppImage**
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

The behavior is still the same - no visible reaction to attribute Line Width in TechDraw.

luzpaz, 2024-04-17:
Is it just me or does the PPA not show `PPA` in the 'About' in ?

WandererFan, 2024-05-17:
Isn't this a duplicate of #6698?

adrianinsaval, 2024-07-21:
I assume this means that the issue is present only on the stable ppa, is the daily ppa affected?

luzpaz, 2024-08-27:
@ajohn77 can you test the daily PPA ?

softlab, 2024-09-12:
Freecad RC1,  Kubuntu 22.04. Daily ppa:freecad-maintainers/freecad-daily),  Viruald box VM
[image]
Export looks good. 

BTW, could you set the default collor of objects into added  views to white

WandererFan, 2024-09-21:
<<BTW, could you set the default collor of objects into added views to white>>

I don't know where the change to blue (turquoise?) for the default face colour came from, but you should be able to set it back to white on the Colors tab of Preferences.

[image]

WandererFan, 2024-10-19:
The default turquoise colour has been corrected.  

The ppa issue is covered here: https://github.com/FreeCAD/FreeCAD/issues/6698

closed 2024-10-19 by WandererFan