FreeCAD/FreeCAD #13478: The application "FreeCAD" can't be opened.

ok-coder1, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I downloaded FreeCAD using Homebrew using `brew install freecad`. After installing, when I try to run it, I get this error (please refer to #13452). This error shows up only when running it through from Spotlight. When I run it from the Applications folder by double-clicking it, it just opens for less than a second and closes.

Log when running from binary:
```shell
Users-iMac:~ User$ /Applications/FreeCAD.app/Contents/MacOS/FreeCAD ; exit;

LIST OF PACKAGES:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.1           py310hb372a2b_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.7.1                h93d8f39_0    conda-forge
arpack                    3.8.0           nompi_hb44a6d1_101    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
blas                      2.120                  openblas    conda-forge
blas-devel                3.9.0           20_osx64_openblas    conda-forge
blinker                   1.7.0              pyhd8ed1ab_0    conda-forge
blosc                     1.21.5               heccf04b_0    conda-forge
brotli                    1.1.0                h0dc2134_1    conda-forge
brotli-bin                1.1.0                h0dc2134_1    conda-forge
brotli-python             1.1.0           py310h9e9d8ca_1    conda-forge
bzip2                     1.0.8                h10d778d_5    conda-forge
c-ares                    1.24.0               h10d778d_0    conda-forge
ca-certificates           2023.11.17           h8857fd0_0    conda-forge
cairo                     1.18.0               h99e66fa_0    conda-forge
calculix                  2.21                 h1235b64_1    conda-forge
certifi                   2023.11.17         pyhd8ed1ab_0    conda-forge
cgal-cpp                  5.6                  h421f912_1    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.2                h25631fb_1    conda-forge
contourpy                 1.2.0           py310ha697434_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                h0dc2134_0    conda-forge
docutils                  0.20.1          py310h2ec42d9_3    conda-forge
double-conversion         3.3.0                he965462_0    conda-forge
eigen                     3.4.0                h1c7c39f_0    conda-forge
expat                     2.5.0                hf0c8a7f_1    conda-forge
ffmpeg                    6.1.0           gpl_h0a522cb_102    conda-forge
flann                     1.9.2                h3d5753d_0    conda-forge
fltk                      1.3.9                he73b29e_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.46.0          py310hb372a2b_0    conda-forge
freecad                   0.21.2          py310h81176b3_0    freecad
freeimage                 3.18.0              h42008e6_18    conda-forge
freetype                  2.12.1               h60636b9_2    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
frozenlist                1.4.1           py310hb372a2b_0    conda-forge
gdk-pixbuf                2.42.10              hbb5a27d_4    conda-forge
gettext                   0.21.1               h8a4c099_0    conda-forge
giflib                    5.2.1                hb7f2c08_3    conda-forge
gl2ps                     1.4.2                h4cff582_0    conda-forge
glew                      2.1.0                h046ec9c_2    conda-forge
glib                      2.78.3               hf4d7fad_0    conda-forge
glib-tools                2.78.3               hf4d7fad_0    conda-forge
gmp                       6.3.0                h93d8f39_0    conda-forge
gmpy2                     2.1.2           py310hb691cb2_1    conda-forge
[... 227 lines cut ...]
3D parametric modeller
==> Artifacts
FreeCAD.app (App)
==> Analytics
install: 536 (30 days), 1,774 (90 days), 7,709 (365 days)
```

### Full version info

```shell
FreeCAD version downloaded: 0.21.2
macOS version: 10.12 Sierra
```

### Subproject(s) affected?

None

### Anything else?

#13452 - GitHub Discussions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ok-coder1, 2024-04-17:
Anybody? I might just change to some other CAD software.

luzpaz, 2024-04-20:
Please connect with other macOS users in the FreeCAD forum. macOS is problematic across the whole FLOSS spectrum.

jbaehr, 2024-05-22:
> macOS version: 10.12 Sierra

This is really old. It has seen its lasst security fix nearly five years ago (around the same time as Ubuntu 14.04). My guess is that the freecad binaries you're trying simply require a more recent version of macOS.

I propose to close this issue as "wont fix".

ok-coder1, 2024-05-23:
[image]

ok-coder1, 2024-05-23:
Issue continuing on forum https://forum.freecad.org/viewtopic.php?p=756779#p756779

ok-coder1, 2024-08-26:
For anyone who comes across this,
I am using 0.20 now and it works. Please check the forum.

closed 2024-08-26 by ok-coder1