FreeCAD/FreeCAD #13481: Core: CopyOnChangeGroup is inconsistently referenced in App::Part parent object.

brittanyb, 2024-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Issue**:

When an ``App::Link`` to an object is created, and the 'Link Copy On Change' property behaviour set to 'Owned', this is managed through a generated ``App::LinkGroup`` called 'CopyOnChangeGroup' (defined in ``src/app/Link.cpp``).

If the link is added to a 'Part' object, and the 'Link Copy On change' property then set to 'Owned' after being moved, the user will receive the message:

```
App::Link: Link(s) to object(s) 'CopyOnChangeGroup' go out of the allowed scope 'Link'. Instead, the linked object(s) reside within 'N/A'.
```

This issue is not present if the link is placed within an ``App::DocumentObjectGroup`` object. The ``App::Part`` appears to need to directly reference the 'CopyOnChangeGroup' object within its 'Group' link list property to prevent the errors occurrence, whereas ``App::DocumentObjectGroup`` does not require this link list reference.

This issue will affect variant part links, and be especially prevalent in the upcoming Assembly workbench.

**Current resolution**:

The user can move the link outside of the parent object, to place it at the root of the document. Then, when re-added to the parent object, the 'Group' property of the parent will be updated with both the created 'Link', and the Link's child 'CopyOnChangeGroup' object which prevents the error message popping up. The user can also manually add the 'CopyOnChangeGroup' into the parent object's 'Group' list property, or set up the 'Link' and its 'CopyOnChange' behaviour in the parent before moving it into a child.

**Bug reproduction steps**:

1. Create two new documents, 'Assembly' and 'AsmPart'.
2. In the 'AsmPart', create an ``App::Part`` object, with a ``PartDesign::Body`` child. Populate the solid however you wish. Save this document.
3. In 'Assembly', create an ``App::Part`` object.
4. In 'Assembly', Create an ``App::Link`` object, move it to be a child of the ``App::Part`` root object.
5. Set the ``App::Link`` 'LinkedObject' property to the 'Part' object of the 'AsmPart' document.
6. Set the ``App::Link`` 'LinkCopyOnChange' property to 'Owned'.
7. Trigger 'Refresh' and 'Re-compute' operations on the document to observe the persistent error within the Report View.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.12 (Disabled)
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * FeedsAndSpeeds 0.5.0
  * parts_library
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-04-20:
@brittanyb are you able to test @realthunder's proposed solution? If so, can you comment on the PR? Thanks!

brittanyb, 2024-04-20:
> @brittanyb are you able to test @realthunder's proposed solution? If so, can you comment on the PR? Thanks!

Will take a look tomorrow, thanks :)

closed 2024-04-24 by chennes