FreeCAD/FreeCAD #13487: Spreadsheet: `importXLSX.py` "some cells don't have `t` and `s` attributes"

luzpaz, 2024-04-15:
https://github.com/FreeCAD/FreeCAD/blob/4f4df894422e1a25e9a236fa43746bd70a215fca/src/Mod/Spreadsheet/importXLSX.py#L335

julhas786, 2024-04-16:
The reported issue in importXLSX.py occurs when encountering cells in the XLSX file that lack both "t" (type) and "s" (style) attributes. This can lead to incorrect parsing and data import into the FreeCAD spreadsheet.

Here's how we can analyze and approach a solution:

**1. Understanding the Problem:**

  **Missing Attributes:** The 't' attribute defines the cell type (e.g., 'n' for number, 's' for string), while the 's' attribute refers to the cell style.
  **Impact:** Without these attributes, the importer cannot correctly interpret the cell's content and format.
  **Possible Causes:** This might occur due to:
    **XLSX File Corruption:** The XLSX file itself may be corrupted or have formatting issues.
    **Specific Excel Features:** The file might utilize features not fully supported by the importer, leading to missing attributes.
    **Parser Limitations:** The current parser implementation may not handle all possible cell formats.

**2. Potential Solutions:**

Here are some potential approaches to address this issue:

**A. Enhance the Parser:**

  **Default Type:** If no 't' attribute is found, assume a default type (e.g., string) and attempt to parse the cell value accordingly.
  **Style Handling:** Investigate how the 's' attribute affects parsing and consider implementing basic style interpretation.
  **Error Handling:** Implement better error handling for cells with missing attributes. This could involve logging a warning or skipping the cell import with a notification.

**B. Investigate XLSX File and Excel Features:**

  **File Validation:** Check the XLSX file for corruption or formatting issues using dedicated tools.
  **Feature Support:** Identify specific Excel features that might lead to missing attributes and consider expanding support for them in the importer.
**C. User Feedback and Fallback Options:**
  **User Notification:** Inform the user about cells with missing attributes and potential data loss.
  **Manual Correction:** Provide options for the user to manually specify the cell type and format after import.
  **Alternative Formats:** If the XLSX format proves problematic, consider supporting alternative spreadsheet formats (e.g., CSV) for import.

**3. Implementation:**

  The specific implementation will depend on the chosen approach and the complexity of the issue. It may involve modifying the `handleCells `function in importXLSX.py to include additional checks and logic for handling cells with missing attributes.

**4. Testing and Validation:**

Thorough testing with various XLSX files and Excel features is crucial to ensure the effectiveness of the implemented solution.

**Additional Considerations:**

  **Performance Impact:** Evaluate the performance impact of any changes to the parser.
  **Backward Compatibility:** Consider maintaining backward compatibility with existing XLSX files.
  **Future-Proofing:** Design the solution with future Excel features and potential format changes in mind.

Roy-043, 2024-04-16:
@luzpaz: I am not sure if it is wise to create GH issues for all occurrences of "todo" and "fixme" in the code.

luzpaz, 2024-04-16:
Roy, I'm trying to be discerning and therefore just choosing ones that seem important and 'overlooked'.

JonasVgt, 2024-04-26:
I have done some digging on this issue.

This part of the code retrieves the type for a specific cell in the spreadsheet, which is defined by the "t" attribute. As defined in the schema, this attribute is optional with a default value of "n":

`<xsd:attribute name="t" type="ST_CellType" use="optional" default="n"/>`

The comment probably refers to the special case of this "t" attribute not being present. In my opinion, the code handles this case perfectly fine, so the comment can probably just be removed.

closed 2024-09-19 by chennes

luzpaz, 2024-09-20:
@JonasVgt thanks again for your feedback and helping to close this ticket. Any interest in hitting some of the other spreadsheet tickets ? https://github.com/FreeCAD/FreeCAD/labels/Mod%3A%20Spreadsheet

@hyarion thanks for making the PR :rocket: