FreeCAD/FreeCAD #13488: Reading and writing to spreadsheet that has color applied to all cells freezes for a multiple minutes

ansonl, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad freezes when selecting all cells and setting a color for it. I need to set the color to another dark color for all cells because there seems to be a bug where the Black text color actually appears as White text color in the cell. 

After setting a text color to all the cells, saving and reopening the Freecad project file and opening the spreadsheet hangs for multiple minutes. The CPU usage of the program appears stuck at 100% for a single thread during this time so it looks like it is trying to process too many cells that have color set to them even though the cells are off screen and have no data set. 

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.2
  * kicadStepUpMod 10.22.1
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Attached is a sample project with a single spreadsheet where all cells have the color previously set to dark blue color. Opening the project and opening the spreadsheet takes minutes and saving the file can take even longer. 
[tire-screw-2piece.zip](https://github.com/FreeCAD/FreeCAD/files/14986657/tire-screw-2piece.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-16:
Are you selecting with Ctrl + A?
Could you also test this in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-04-16:
I can reproduce on the currently weekly. Even selecting all takes a few seconds.
I guess it will read the values of all cells and changes them which is huge.

ansonl, 2024-04-16:
Yes, I was selecting on with Ctrl + A and the same results happen.

xtemp09, 2024-04-16:
This is a duplicate of #11335.

maxwxyz, 2024-04-16:
Duplicate of #11335

maxwxyz, 2024-04-16:
@xtemp09 thanks!

closed 2024-04-16 by maxwxyz