FreeCAD/FreeCAD #13490: Part naming not unique in Elmer SIF files / Not all boundary conditions written

xorxx, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My Sif files created for elmer solver often don't execute because of error message:
```
ReadTargetNames: Reading names info from file: ./mesh.names
ERROR:: ReadTargetNames: Mapping of name is not unique: solid1
```
For multiple solids or other object types only the name part "solid" is written to the sif file. If you change it at one position in the file to an other name e.g. "solid1b" the file is running. I guess a complete object identifier should be used as a name.

For the attached project the naming problem is not the only one. I have three boundary conditions but only two are written to sif file.

The condition "ConstraintElectrostaticPotential-negative" is not written. When I remove it and create it from scratch it is written but another condition is missing then. Maybe a one-off problem?

Thanks for your support and for this great software!

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * freecad.gears 1.1.0
  * Help 1.0.3
  * ToOptix
[/code]
```

### Subproject(s) affected?

FEM

### Anything else?

1. Freecad project with two problems in SIF file:  [FEM-Elmer-Problem.zip](https://github.com/FreeCAD/FreeCAD/files/14996421/FEM-Elmer-Problem.zip)
2. Files written for Elmer Solver: [FEM-Elmer-problem-sif.zip](https://github.com/FreeCAD/FreeCAD/files/14996438/FEM-Elmer-problem-sif.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-16:
@FEA-eng FYI

FEA-eng, 2024-04-17:
I'm not familiar with this part of the code. @marioalexis84 Can you take a look at it?

xorxx, 2024-04-29:
I did some debugging on this problem, but I am not deep enough in code to provide a final solution. But some hints from my side.

At least a part of the problem is inside `.\Mod\Fem\femsolver\elmer\writer.py`. At several places in code a name for the sif-file name-field is created by only using `name `property of objects. This leads to duplicate because `name `property has values like "solid1" or "face1" for multiple objects. I tried to resolve this. Therefore I modified function getAllBodies as follows by replacing references to `obj.Name` with `obj.FullName`. I added a string replace for "#" because this leads to problem in SIF file.

```
 def getAllBodies(self):
     obj = self.getSingleMember("Fem::FemMeshObject")
     bodyCount = 0
     prefix = ""
     if obj.Part.Shape.Solids:
         prefix = "Solid:"+obj.FullName.replace("#",":")
         bodyCount = len(obj.Part.Shape.Solids)
     elif obj.Part.Shape.Faces:
         prefix = "Face:"+obj.FullName.replace("#",":")
         bodyCount = len(obj.Part.Shape.Faces)
     elif obj.Part.Shape.Edges:
         prefix = "Edge:"+obj.FullName.replace("#",":")
         bodyCount = len(obj.Part.Shape.Edges)
     return [prefix + str(i + 1) for i in range(bodyCount)]
```

Modifications like this are needed in other places too. I am just not familiar enough with SIF file structure to see if this is really correct what I am doing here. At least my model is not fully running as expected with this modifications, but it gets a bit better.

Maybe this helps one of the specialist by solving this problem. Thanks for all your contributions to this great piece of software! I convinced by boss to donate some money to FreeCAD development. It is definitely worth!

marioalexis84, 2024-05-20:
I will have a look

artsonit, 2024-08-01:
Hi, 
change line 189 of .\Mod\Fem\femsolver\elmer\sifio.py
 
 section["Name"] = name 

to 

 section["Name"] = name + "x"

luzpaz, 2024-08-02:
@artsonit if this is the fix, can you make a PR ?

akapelrud, 2024-09-04:
Bumping this. I had a similar issue and tried the proposed solution. It fixed half of the problem.:

> Hi, change line 189 of .\Mod\Fem\femsolver\elmer\sifio.py
> 
> section["Name"] = name
> 
> to
> 
> section["Name"] = name + "x"

This removes the error by changing the names of the exported boundary conditions, but it does not fix all the elmer output files. Specifically, the generated file `mesh.names` must be altered to reflect the same changes in the section:
`! ----- names for boundaries -----`.

luzpaz, 2024-09-04:
Where does the `mesh.names` get generated from and separately question: how do other installed solvers do this. Can someone compare ?

akapelrud, 2024-09-06:
> Where does the `mesh.names` get generated from and separately question: how do other installed solvers do this. Can someone compare ?
>
I am pretty sure that it is `elmergrid` that generates this file. It is probably fed the names by the writer somehow.

luzpaz, 2024-09-06:
We need a dev to spearhead this effort. If you know someone or of someone that could, please pass the word on.

akapelrud, 2024-09-09:
The boundary conditions are named in the input file `mesh.unv` generated in in `src/Mod/Fem/femsolver/elmer/writer.py` which is fed to elmergrid. It seems that the lines:

```python
    def _writeMesh(self):
        mesh = self.getSingleMember("Fem::FemMeshObject")
        unvPath = os.path.join(self.directory, "mesh.unv")
        groups = []
        groups.extend(self._builder.getBodyNames())
        groups.extend(self._builder.getBoundaryNames())
        self._exportToUnv(groups, mesh, unvPath)
        ...
```
I'll guess this can be fixed by making sure that `getBodyNames()` and `getBoundaryNames()` create unique names. Maybe add a 'BC_' prefix to the entries from `getBoundaryNames()`

xorxx, 2025-06-24:
Issue still persists with weekly build 42180 on Windows.