FreeCAD/FreeCAD #13492: Continuous trimming doesn't work

prokoudine, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Continuous trimming doesn't work for me:

1. Select Trim tool
2. Press and hold left mouse button
3. Start dragging over line segments
4. No segments get deleted

https://github.com/FreeCAD/FreeCAD/assets/57467/91d257ac-2281-467e-ac8f-5c7eaa9f372b

I get the same with at least CAD, Blender, and Touchpad navigation styles, if that matters.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git) AppImage
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * ExplodedAssembly
  * sheetmetal 0.3.15
  * BIM 2021.12.0
  * A2plus 0.4.61
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2023.12.13.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-04-16:
I cannot reproduce this problem so it is hard for me to find the root cause. Is there anyone else for whom it doesn't work ?

PaddleStroke, 2024-04-16:
Ah yes it seems the commit hash in your version number is older than the commit of the trim tool (4b968a607de5a86744fd5e421b2f6485e330ea0d)

That would certainly explain why it's not working ;)

maxwxyz, 2024-04-16:
Can I close this then?

FEA-eng, 2024-04-17:
> Can I close this then?

It's available in today's build (36874) so yeah, the issue can be closed.

closed 2024-04-17 by maxwxyz