FreeCAD/FreeCAD #13493: Failed to add geometry: index out of bound

YosypC, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/50027752/af417df4-e2ad-4892-95dd-6abf42b594a1

The whole file seems to behave very strangely. I didn't catch on video, but Sketch002 was once opened with absolute no constraints applied: a quick Undo and Redo "restored" it. I have no way of breaking it again.
If you go back some steps, like 10, and try opening it, it contains errors and asks for the validation tool, but the latest iterations are good, or so it seems.

The vertices I am trying to project into the sketch have already been projected in the past, but after some modification in the original sketch, they've been somehow... lost? But it didn't prompt me with descynced references in Sketch002, they just disappeared.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36789 (Git)
Build type: Unknown
Branch: main
Hash: ff2bca2505010f580c9c35abfdfe8507e1c0d1cb
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

None

### Anything else?

[NewSpeedloader (copy).zip](https://github.com/FreeCAD/FreeCAD/files/14999154/NewSpeedloader.copy.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-08:
@YosypC is this still an issue in the latest weekly?

YosypC, 2024-11-08:
@maxwxyz I am unable to test anything for the upcoming months. I might be able to cut half an hour of my time but I unfortunately can't guarantee.

luzpaz, 2025-06-24:
@YosypC have you run into this on 1.0.x or 1.1.x yet ?

YosypC, 2025-06-24:
@luzpaz I don't think so. I've had limited experience with 1.0 onwards. I guess we can close this.