FreeCAD/FreeCAD #13494: clicking on the navigation cube does nothing

greyltc, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The navigation cube and all associated controls up there in the top right corner of the render view do not respond to clicks.

### Full version info

```shell
[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Intel Arc GPU, Linux 6.8.5, gnome 46.0 desktop, i915 driver.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-16:
Left or right click? Are you using a mouse, touch pad, touchscreeen? 
Can you also try it in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

greyltc, 2024-04-16:
Mouse. Neither, left nor right clicking works. I'll see if I can get a weekly build going and report back shortly.

greyltc, 2024-04-16:
@maxwxyz Okay, I've built and tested whatever commit that latest weekly build you linked there is based on (a4dd600ec6d5a53a9960cd9f8693d7d307b9e359). The cube's behavior has changed in that now the cube and associated controls highlight whenever my pointer hovers over them, but still nothing up there responds in any way to any left or right mouse clicks/click+drag or anything at all. See attached video:
[Screencast from 2024-04-16 11-33-00.webm](https://github.com/FreeCAD/FreeCAD/assets/16778174/b408ee83-1b93-4146-8a9e-1e5464d06b75) I can't quite figure out why/when the cursor decides to change between a hand and an arrow. Right clicking brings up the menu as shown.

```
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: makepkg
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

maxwxyz, 2024-04-16:
I cannot reproduce on my system and never heard about this behavior. Could you try temporarily rename your user.cfg file?

greyltc, 2024-04-16:
`mv ~/.config/FreeCAD/ ~/.config/FreeCAD.bak` then relaunch freecad did not impact the issue.

I've got this issue on multiple machines (desktops and laptops). Granted, these are all pretty similar setups: Arch Linux, gnome desktop environment, intel graphics, freecad from official Arch repos (and now the one test I did where I built from a4dd600ec6d5a53a9960cd9f8693d7d307b9e359)

maxwxyz, 2024-04-16:
@greyltc related to this one? https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/3

adrianinsaval, 2024-04-22:
I consider this issue to be better at describing the navicube issue than the above and would advise to have this one open and close the other one instead.

@Rexbas would you have any clue why the navicube doesn't properly register mouse events when built with qt6?

Rexbas, 2024-04-23:
I have no idea but I am trying to compile with qt6 right now.

Rexbas, 2024-04-23:
It seems that the behavior of `glReadPixels(...)` is different but I don't know why or how to fix it.

https://github.com/FreeCAD/FreeCAD/blob/eb3e83b97a345620e0e26f3051d3d7a3f26755c6/src/Gui/NaviCube.cpp#L936-L937

adrianinsaval, 2024-04-24:
could it be related to dpi scaling?

Rexbas, 2024-04-24:
Don't know if it is dpi scaling.

If I write the `QtGLFramebufferObject` to a file by inserting this code:
```
auto img = m_PickingFramebuffer->toImage();
img.save(QString::fromStdString("test.png"));
```
here: https://github.com/FreeCAD/FreeCAD/blob/eb3e83b97a345620e0e26f3051d3d7a3f26755c6/src/Gui/NaviCube.cpp#L930

Then with qt5 I get this:
[image]
It might be hard to see but it is the navigation cube in dark red on a black background. This image is 800x800 pixels when I set navigation cube size to 400.

But with qt6 I get this:
[image]
This is also a 800x800 image but the content of the image is clearly not right.

So I think something is wrong with `QtGLFramebufferObject* m_PickingFramebuffer`. I don't know how to fix.

YakoYakoYokuYoku, 2024-05-12:
I'm currently investigating this issue, from my observations this is split in two problems. One is that there's no mouse interaction with the navigation cube, so this means that at least one of SoQt or Qt6 is the culprit. The other is that reading from the framebuffer object returns invalid data, @Rexbas from what I can check is that `QOpenGLFramebufferObject` [hasn't changed its semantics essentially since Qt5](https://invent.kde.org/qt/qt/qtbase/-/commits/dev/src/opengl/qopenglframebufferobject.cpp?ref_type=heads), so it must another thing.

YakoYakoYokuYoku, 2024-05-12:
Could be related coin3d/soqt#80.

Rexbas, 2024-05-12:
I think it is one problem. The mouse interaction with the navigation cube is not working because it can't decide which face is hovered/selected in `NaviCubeImplementation::pickFace` which is caused by the different behavior in the frame buffer.

YakoYakoYokuYoku, 2024-05-19:
Good news @Rexbas! I've found the reason why this happens. It turns out that what Qt has done might be correct and we are doing things wrong. By calling `glGetError` in `gdb` I've noticed that when we draw the cube faces in `NaviCubeImplementation::drawNaviCube` an OpenGL stack overflow occurs, hence why you're getting a corrupted read on the framebuffer object.

https://github.com/FreeCAD/FreeCAD/blob/eb3e83b97a345620e0e26f3051d3d7a3f26755c6/src/Gui/NaviCube.cpp#L842-L857

The only culprit I can point out is `glVertexPointer`, as it was when that function is called that `GL_STACK_OVERFLOW` is produced plus it was deprecated and not included with other fixed functions with OpenGL 2.0 and onwards. Exactly why this happens with Qt6, I don't know, but I'm certain that this is due to `glVertexPointer`. The solution would be to use something else like `glVertexAttribPointer` or VBOs.

PR #14159 (2024-05-22): Gui: Fix NaviCube drawing and interaction with Qt6
I've noticed that while debugging the NaviCube with a FreeCAD Qt6 build OpenGL errors are produced when picking faces or buttons. It turns out that these errors are produced because the underlying framebuffer object has a different context than the one that was created with, more specifically, QtWebEngine changes it before the mouse event is processed. I've fixed this by setting making the proper context current explicitly at the 3D Inventor viewer when Navicube has to process the event. Please test this and give your comments before merging.  [... cut]
closed 2024-05-22 by adrianinsaval