FreeCAD/FreeCAD #13497: GCode export for RepRapFirmware fails

giedrius-stanevicius, 2024-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If exporting multiple files with `rrf` post processor, only first file export succeeds and then error is logged in console.

It seems that `export` method  in `rrf_post.py` is missing `return final` at the end.

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-16:
Could you also try the same file in the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
TIA!

giedrius-stanevicius, 2024-04-16:
@maxwxyz Yes, it's the same with dev build

luzpaz, 2024-07-12:
Is this a regression ?

sslupsky, 2024-12-05:
Might be related.  There is a problem saving the file using the rrf post processor in Build 39380:

```
[Errno 30] Read-only file system: '-'Running the Python command 'CAM_Post' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD 1.1.0dev.39380.app/Contents/Resources/Mod/CAM/Path/Post/Command.py", line 188, in Activated
    post_data = postprocessor.export()
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 1.1.0dev.39380.app/Contents/Resources/Mod/CAM/Path/Post/Processor.py", line 321, in export
    gcode = self.script_module.export(sublist, "-", self._job.PostProcessorArgs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 1.1.0dev.39380.app/Contents/Resources/Mod/CAM/Path/Post/scripts/rrf_post.py", line 451, in export
    with open(filename, "w") as fp:
         ^^^^^^^^^^^^^^^^^^^

[Errno 30] Read-only file system: '-'
```

IIRC, there were some changes made and this post processor was not updated?  Most post processors use a different method for writing the file now as follows:
https://github.com/FreeCAD/FreeCAD/blob/760091bc8ad1f96862d01bea3fad642834f8971a/src/Mod/CAM/Path/Post/scripts/linuxcnc_post.py#L273-L278

OS: macOS 14.6.1
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default

blaggacao, 2025-07-13:
The return signature of the rff post processor does not match the expectation within `Mod/CAM/Path/Post/Command.py` 

```python

def Activated(self):

    ...
    post_data = postprocessor.export()
    ...
    for item in post_data:
        subpart, gcode = item
        ...
```

Whereas rff's export has a `None` type return signature.

The result is:

```
18:49:01  Running the Python command 'CAM_Post' failed:
Traceback (most recent call last):
  File "......../Mod/CAM/Path/Post/Command.py", line 207, in Activated
    self._write_file(fname, gcode, policy)
  File "......../Mod/CAM/Path/Post/Command.py", line 135, in _write_file
    f.write(gcode)

write() argument must be str, not None
```

blaggacao, 2025-07-13:
Absolutely! (sry)

https://github.com/FreeCAD/FreeCAD/blob/566fe8371a88938e1b70a6048caac09f345cf024/src/Mod/CAM/Path/Post/Command.py#L216-L228

While looking, maybe this fixes it on the rrf side: https://github.com/FreeCAD/FreeCAD/commit/1809c6cf01cfe28e4f2abe1c31c0bf6c19e1ef76

It now returns `final`, but I can't see how this is actually a list of tuples. It looks more like a list of strings/bytes to me. Maybe there are hidden (from obvious code path) type transformations going on that make this work?

tarman3, 2025-07-13:
Can not reproduce issue

[helix.zip](https://github.com/user-attachments/files/21206005/helix.zip)

blaggacao, 2025-07-13:
@tarman3 My understanding of the issue is such, that it is not dependent on any particular FCStd. Please teach me otherwise, if you're confident that a particular FCStd would reproduce this.

Witch strong confidence from the function interface mismatch, I can state that the only relevance to the issue is that the particular code path is triggered. In (released) `v1.0.0`, this is the case, when you click on the "Post Process" icon next to the create job icon, labeled as _CAM_Post_.

tarman3, 2025-07-13:
Two files saved
No any errors

https://github.com/user-attachments/assets/c352aea9-b5a5-4271-a362-0623bbd68dd8

blaggacao, 2025-07-13:
Did you check that the `nc` file is not actually empty? (which is the case for me)

If it's not empty, that can only mean that `1.1.0dev.42535` apparently fixes this and something non-obvious happens with the `final` return type.

tarman3, 2025-07-13:
nc files is ok

blaggacao, 2025-07-13:
Looks like this might be closed on the next point release, then 👍

blaggacao, 2025-07-18:
This can be closed @maxwxyz:

> It seems that export method in rrf_post.py is missing return final at the end.

Implemented by https://github.com/FreeCAD/FreeCAD/commit/1809c6cf01cfe28e4f2abe1c31c0bf6c19e1ef76

Validated by @tarman3 and myself on `weekly-2025.07.14`