FreeCAD/FreeCAD #13505: Assembly: Loading WB crashes FreeCAD instantly

maxwxyz, 2024-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On a fresh install, no mods or old preference files with the current weekly Windows  24-04-17, FreeCAD crashes instantly when trying to switch to the Assembly WB or loading the WB via the preferences.
@PaddleStroke FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-04-17:
Having read this, I tried to reproduce. No crash here:

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

PaddleStroke, 2024-04-17:
Is this recent ?
The only PR that merged on assembly is https://github.com/FreeCAD/FreeCAD/pull/13245 so perhaps it's this one.

Are you opening a file or is it an empty assembly? If you have an invalid joint, perhaps its throwing the error and stopping freecad.
This PR 13245 is already overriden in my branch anyway

maxwxyz, 2024-04-17:
No file open. Fresh install, no prior FreeCAD on that PC. Just opening the application and switching to the assembly WB crashes

xtemp09, 2024-04-17:
@maxwxyz, could you provide the backtrace?

maxwxyz, 2024-04-17:
No, sorry

xtemp09, 2024-04-17:
@maxwxyz, could you provide also the _PATH_ environment variable?

maxwxyz, 2024-04-17:
> @maxwxyz, could you provide the _PATH_ environment variable?

How, what's this?

maxwxyz, 2024-04-17:
@xtemp09 FreeCAD is not in the path variable. I have no FreeCAD installed, just downloaded the weekly build and ran it.

xtemp09, 2024-04-17:
@maxwxyz, could you just provide the _PATH_ variable?

maxwxyz, 2024-04-17:
I cannot for this PC, sorry. Is there another way to narrow it down or provide more info?

maxwxyz, 2024-04-18:
Unfortunately it needs admin privileges on this computer, so also not feasible for me

maxwxyz, 2024-04-18:
How can I record the log file?

xtemp09, 2024-04-18:
https://wiki.freecad.org/Debugging

maxwxyz, 2024-04-22:
Issue still persists with fresh weekly build, deleted %APPDATA%/FreeCAD folder
I cannot share the log file from this computer.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git)
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.34
  * fasteners 0.5.17
  * OpenTheme 2024.4.20
  * OpticsWorkbench 1.0.16
  * sheetmetal 0.4.11
```

xtemp09, 2024-04-22:
@maxwxyz, you don't provide:

1. The log file
2. The backtrace
3. Specs
4. %_PATH_%
5. Anything else

How can you possibly imagine the solution?

maxwxyz, 2024-04-22:
@xtemp09 
here is a cleaned up logfile, I've deleted the local paths:

<details><summary>Details</summary>
<p>

```
Msg: FreeCAD 0.22.0, Libs: 0.22.0devR36874 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Time = 2024-Apr-19 18:58:59
Log: AboutImage = freecadabout
Log: AppDataSkipVendor = true
Log: AppHomePath = FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\
Log: AppIcon = freecad
Log: AppTempPath = (deleted)
Log: BOOST_VERSION = 1_82
Log: BinPath = FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\bin\
Log: BuildRepositoryURL = https://github.com/FreeCAD/FreeCAD main
Log: BuildRevision = 36874 (Git)
Log: BuildRevisionBranch = main
Log: BuildRevisionDate = 2024/04/16 04:00:27
Log: BuildRevisionHash = a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 22
Log: BuildVersionPoint = 0
Log: BuildVersionSuffix = dev
Log: Console = 0
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Debug = 0
Log: DocPath = FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\doc\
Log: EIGEN_VERSION = 3.4.0
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.22.0
Log: LoggingConsole = 1
Log: LoggingFile = 1
Log: LoggingFileName = (deleted)\AppData\Roaming\FreeCAD\FreeCAD.log
Log: MaintainerUrl = https://www.freecad.org/wiki/Main_Page
Log: OCC_VERSION = 7.7.2
Log: PATH = (deleted)
Log: PYSIDE_VERSION = 5.15.8
Log: PYTHON_VERSION = 3.11.8
Log: PythonSearchPath = FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\bin\python311.zip;FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\bin\DLLs;FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\bin\Lib;FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\bin
Log: QT_VERSION = 5.15.13
Log: RunMode = Gui
Log: SMESH_VERSION = 9.8.0.2
Log: SplashAlignment = Bottom|Left
Log: SplashInfoColor = #8aadf4
Log: SplashInfoPosition = 6,75
Log: SplashScreen = freecadsplash
[... 119 lines cut ...]
Log: All modules with GUIs initialized using pkgutil are now initialized
Log: Init: Loading FreeCAD GUI
Log: Init: Running FreeCADGuiInit.py start script... done
Log: Init: Activating default workbench StartWorkbench
Log: Loading Material module... done
Log: Module: Part
Log: Loading Part module... done
Log: Loading Sketcher module... done
Log: Loading PartDesign module... done
Log: Loading GUI of Material module... done
Log: Loading GUI of Part module... done
Log: Loading GUI of Sketcher module... done
Log: Loading GUI of PartDesign module... done
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: Init: Entering event loop
Log: Init: Executing event loop...
Log: Init: Processing command line files

``` 

</p>
</details>

xtemp09, 2024-04-22:
The whole point is to know %_PATH_%. Both Qt and Python are known to load plugins (modules) from other directories. Incompatibility between different versions cause trouble.

maxwxyz, 2024-04-22:
@xtemp09 yes, it launches but still crashes with the clean path as linked.
The log file says `Log: PATH = "C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;"`

PaddleStroke, 2024-05-10:
Is this still an issue since the merge of 12406?

maxwxyz, 2024-05-11:
no, I'll close this.

closed 2024-05-11 by maxwxyz