FreeCAD/FreeCAD #13510: Crash on sketch loading

RiccaDS, 2024-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the file I attach when I double click on Sketch011 of the "base_valvola" part in the tree view, FC crashes. The sketch is never displayed and then it takes some 5-10-15 seconds but it crashes. 

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36677 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fadb1f40bdfa4ee1c61acbf12fbe44680723965
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.10
  * NikraDAP 2.0.0
  * Render 2024.1.23
  * Assembly4 0.50.12
  * fasteners 0.5.15
  * workfeature-macro
  * CurvedShapes 1.0.8
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Curves 0.6.29
  * OSE3dPrinter 0.1.0
  * Design456 0.0.1
  * kicadStepUpMod 11.1.0
  * QuickMeasure 2022.10.28
  * pcb 6.2023.1
  * Launcher
  * ExplodedAssembly
  * toSketch 1.0.1
  * A2plus 0.4.64
  * Manipulator 1.5.7
  * MakerWorkbench 1.0.1
  * Estimate 0.1.2
  * workfeature
  * FEMbyGEN 2.1.0
```

### Subproject(s) affected?

None

### Anything else?

[image]

[crash.zip](https://github.com/FreeCAD/FreeCAD/files/15011772/crash.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-17:
Sketch won't open on my side, no crash though.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

RiccaDS, 2024-04-17:
> Sketch won't open on my side, no crash though.
>
> ```
> OS: Windows 10 build 19045
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.36874 (Git)
> Build type: Release
> Branch: main
> Hash: [...]

I noticed it crashes if you zoom in/out/pan the in the view

FEA-eng, 2024-04-17:
> Sketch won't open on my side, no crash though.

Yeah, it doesn't open (but shows some Sketcher constraints and elements in the Tasks tab) and doesn't crash on Windows in 0.22 weekly. Just throws a bunch of errors:

```
19:10:55  <Exception> TopoShapeExpansion.cpp(2066): Shape index 71 out of bound 19
19:10:56  <Exception> Access violation
19:10:56  0 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:56  <Exception> Access violation
19:10:56  0.0131486 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:56  <Exception> Access violation
19:10:56  0.0219771 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:56  <Exception> Access violation
19:10:56  0.0300945 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:56  <Exception> Access violation
19:10:56  0.0760465 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:56  <Exception> Access violation
19:10:56  0.0937659 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.335471 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.360978 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.372449 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.426199 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.446323 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.461027 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.625094 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:10:57  <Exception> Access violation
19:10:57  0.641805 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:11:07  <Exception> Access violation
19:11:07  11.0817 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
19:11:09  <Exception> Access violation
19:11:09  12.3683 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
```

xtemp09, 2024-04-17:
_FreeCAD_ does not crash when opening the sketch, but crashes when closing. `SketcherGui::EditModeGeometryCoinManager::updateGeometryColor` again causes the crash.

_FreeCAD_ reports when opening:

<details>

```cpp
01:40:07  pyException: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'
01:40:07  pyException: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'
01:40:43  <Exception> TopoShapeExpansion.cpp(2065): Part::TopoShape Part::TopoShape::getSubTopoShape(TopAbs_ShapeEnum, int, bool) const -- Shape index 71 out of bound 19
01:40:45  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'NameError'>: Unknown document ''
01:40:45  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Unknown document ''
```

</details>

<details><summary>The backtrace</summary>

```cpp
Process 1681 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff50906a18 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
   1123       operator[](size_type __n) _GLIBCXX_NOEXCEPT
   1124       {
   1125         __glibcxx_requires_subscript(__n);
-> 1126         return *(this->_M_impl._M_start + __n);
   1127       }
   1128
   1129       /**
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff50906a18 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
    frame #1: 0x00007fff50906a0d SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] SketcherGui::CoinMapping::getPointGeoId(layerindex=0, pointindex=0, this=0x00005555592aec38) at EditModeCoinManagerParameters.h:488:53
    frame #2: 0x00007fff509069fb SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(this=0x0000555558b07e20, geolistfacade=0x00007fffffffac20, issketchinvalid=false) at EditModeGeometryCoinManager.cpp:193:50
    frame #3: 0x00007fff50901ffc SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x00005555592ae9a0, geolistfacade=0x00007fffffffac20) at EditModeCoinManager.cpp:884:54
    frame #4: 0x00007fff509020bf SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x00005555592ae9a0) at EditModeCoinManager.cpp:877:16
    frame #5: 0x00007fff5091ce47 SketcherGui.so`SketcherGui::ViewProviderSketch::updateColor(this=0x000055555a1ae520) at ViewProviderSketch.cpp:2588:33
    frame #6: 0x00007fff50921540 SketcherGui.so`SketcherGui::ViewProviderSketch::onSelectionChanged(this=0x000055555a1ae520, msg=0x00007fffffffb320) at ViewProviderSketch.cpp:2207:30
    frame #7: 0x00007ffff7187d59 libFreeCADGui.so`Gui::SelectionObserver::_onSelectionChanged(this=<unavailable>, msg=<unavailable>) at Selection.cpp:152:27
    frame #8: 0x00007ffff71966b5 libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=__invoke_memfun_deref @ 0x00007fffffffaf80, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #9: 0x00007ffff71966d3 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] std::__invoke_result<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>::type std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=0x00007fffffffb320, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #10: 0x00007ffff71966ce libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>( [... cut ...]
    frame #95: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #96: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36882 (Git)
Build type: Debug
Branch: main
Hash: cf1eeaab19da2d2a6eaaba3ae6bd1fc8ad620552
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-04-17:
does it work with the stable 0.21.2 version?
Might it be related to the materials update?

RiccaDS, 2024-04-17:
> does it work with the stable 0.21.2 version? Might it be related to the materials update?

tested this now, it doesn't crash in 0.21.2 and sketch opens normally, apparently with no problems until now.
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.10
  * NikraDAP 2.0.0
  * Render 2024.1.23
  * Assembly4 0.50.12
  * fasteners 0.5.15
  * workfeature-macro
  * CurvedShapes 1.0.8
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Curves 0.6.29
  * OSE3dPrinter 0.1.0
  * Design456 0.0.1
  * kicadStepUpMod 11.1.0
  * QuickMeasure 2022.10.28
  * pcb 6.2023.1
  * Launcher
  * ExplodedAssembly
  * toSketch 1.0.1
  * A2plus 0.4.64
  * Manipulator 1.5.7
  * MakerWorkbench 1.0.1
  * Estimate 0.1.2
  * workfeature
  * FEMbyGEN 2.1.0
[/code]

maxwxyz, 2024-04-17:
@PaddleStroke @davesrocketshop 
was`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor` changed with color rendering in sketcher or a materials PR?

davesrocketshop, 2024-04-17:
I'll start looking into this, but I'm also getting this on opening:
`15:28:14  pyException: Traceback (most recent call last):
  File "C:\Users\dcarter\Documents\Projects\freecad-build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'
15:28:14  pyException: Traceback (most recent call last):
  File "C:\Users\dcarter\Documents\Projects\freecad-build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'`

That'll need to be fixed before assuming the error is anywhere else. Does this model use the Fasteners workbench or other external workbench?

davesrocketshop, 2024-04-17:
> I'll start looking into this, but I'm also getting this on opening: `15:28:14 pyException: Traceback (most recent call last): File [...]

Sorry. Just read the configuration from the original post. I'll install fasteners and try again

RiccaDS, 2024-04-17:
> I'll start looking into this, but I'm also getting this on opening: `15:28:14 pyException: Traceback (most recent call last): File [...]

there is a **fasteners** threaded rod in the group "costruzione" but this I don't think is related to the problem as that part is hidden. Just a missing module for you.
As a note, Hole007 is threaded(not displayed) but uses the normal threading option, not the fasteners workbench. However this threading is adjacent to the Groove whose sketch gives the problem. Don't know if this might be related.
Thanks for looking

RiccaDS, 2024-04-17:
I deleted parts from the file to make it simpler and reproduces the problem anyway
[crash1.zip](https://github.com/FreeCAD/FreeCAD/files/15016434/crash1.zip)

PaddleStroke, 2024-04-18:
I downloaded the simplified file and can reproduce the crash : 
`<Exception> TopoShapeExpansion.cpp(2066): Shape index 71 out of bound 19`

updateGeometryColor was changed by me during the sketch colors PR (construction lines and point colors both). However I do not think they could cause the crash. From the error I'm seing, the likely culprit is the current TNP work. @bgbsww @CalligaroV

bgbsww, 2024-04-18:
Reproducible here.  Issue appears to be in SketchObject::restoreFinished which calls SketchObject::validateExternalLinks which ends up down at the error.  Seems to be a case of missing hasher code in TNP work.  More investigation to follow.

RiccaDS, 2024-04-20:
> No, that error is as a result of the file that clearly has a bad external reference in the sketch object. Issue is how it was created, not how it is read back in.

Should not crash tough :D

bgbsww, 2024-04-20:
Agreed on the crash - I'm working on that.  Curious if you have any idea how we came to have a reference to Edge71 and Edge31 in an object with only 19 Edges?  I've tried a lot of things and to create a test to recreate that failure, and I can't so far.

RiccaDS, 2024-04-20:
ehm, no not really, I lost track of what I did. I've been using FreeCAD for quite some time now but I wouldn't consider myself expert in best practices and such. I am aware of the topological naming problem when sketching on the face of a same part but I admit not following the best practice to avoid that. Maybe I sketched, referenced and then modified the base part which resulted in corrupted references. Or maybe the thread modeling has something to do in this. As you can see from the picture, the hole is actually threaded, if I set "Model Thread" to True, I can see edges that have very high numbers (ex Edge67 in pic)
[image]

RiccaDS, 2024-04-20:
> Agreed on the crash - I'm working on that. Curious if you have any idea how we came to have a reference to Edge71 and Edge31 in an object with only 19 Edges? I've tried a lot of things and to create a test to recreate that failure, and I can't so far.
this seems to be Edge71 in fact
[image]

bgbsww, 2024-04-20:
Thanks!  I'll hunt that now.  I do have a working fix for the crash - when the exception is thrown, we leave things in a messy state - I can fix that.  I just don't want to hide the base cause if I can find it too.

RiccaDS, 2024-04-20:
Sure, I made a few tests and can confirm the issue on other similar and simpler models. 
1- Make a threaded hole with "model thread" to True
2- Sketch and reference to one of the edges of the thread
3- Set "model thread" to false 
4- Double click on the sketch and crash happens
 
 Thank you for looking into that.

bgbsww, 2024-04-21:
As far as I can tell, the code that ensures that we correctly handle missing external references applies not just during a file load, but during each close and open of a sketch. So this patch appears to handle both the crash, and the generation of an invalid sketch.  Thanks for the extra detail, RiccaDS.

RiccaDS, 2024-04-24:
> As far as I can tell, the code that ensures that we correctly handle missing external references applies not just during a file load, but during each close and open of a sketch. So this patch [...]

Just checked the weekly build, works great , thank you very much! ideally it would be perfect if the external reference was maintained even when the thread model gets disabled. As I mentioned it's not uncommon in machining setups to be in need of the model with the thread and the one without. So it's pretty frequent to enable/disable the thread model. Correcting this probably however is quite some good work as, in my understanding of how FC manages threaded holes, the thread model geometries are created/deleted each time one enables/disables the thread model, so it would be necessary to implement a "layer" with the thread model, that can be hidden/shown, which I believe can get really not straightforward :D

PR #13582 (2024-05-06): Ensure we set up a valid sketch even with invalid external references
Possible fix #13510.
closed 2024-05-06 by yorikvanhavre