FreeCAD/FreeCAD #13523: Angle constraint in sketcher is inverted from one side to the other of the plane origin

ezequielleonzybert, 2024-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/25140875/66ff2961-5392-40b8-bf89-9478abe340e9

As you can see in the video, the angle constraint changes from one side to the other of the plane origin
I think the correct behavior would be that both sides show the same angle.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Europe (en_150)
Installed mods: 
  * Assembly4 0.50.12
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-18:
This should be solved in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Could you try it and comment?

FEA-eng, 2024-04-18:
> This should be solved in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds Could you try it and comment?

The new Dimension tool doesn't exhibit this behavior but the old Angle constraint still does:

[image]

ezequielleonzybert, 2024-04-19:
Yep, this is solved on dev release. Sorry 

> This should be solved in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds Could you try it and comment?

ezequielleonzybert, 2024-04-19:
Solved in dev release

closed 2024-04-19 by ezequielleonzybert