FreeCAD/FreeCAD #13524: Dragging a sketch around a point inconsistency in sketcher WB

ezequielleonzybert, 2024-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you try to drag some sketch around a point with that sketch being constrained on length, the full turn can't be done if you hold the mouse click all the time, but you can do it by releasing the mouse click and grabbing the sketch again, doing the turn by parts:

https://github.com/FreeCAD/FreeCAD/assets/25140875/e69b7bb9-70ec-4036-9da9-84eb2b3afb5e

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Europe (en_150)
Installed mods: 
  * Assembly4 0.50.12
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-18:
I can confirm that in 0.22 weekly. It may not be a serious bug but it would be nice to fix this.

ezequielleonzybert, 2024-04-18:
> I can confirm that in 0.22 weekly. It may not be a serious bug but it would be nice to fix this behavior.

I know it's not a serious bug but in my case, working in sketcher workbench is almost 100% of my workflow, as I work in simple projects where I just need to test some measures and make a very simple simulation "by dragging the sketches with the mouse"
So even if this is not serious ion the overall, for someone like me, it disturbs a lot.

luzpaz, 2024-04-19:
Heads up @PaddleStroke