FreeCAD/FreeCAD #13527: Toggle Grid button should be a toggle button that reflects the visibility status of the grid

furgo16, 2024-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Draft workbench, the <kbd>Toggle Grid</kbd> toolbar button (`Draft_ToggleGrid`) is a generic click button that toggles the visibility of the grid on or off. However, it does not offer any indication of that status.

A usability improvement would be for the button to be an actual toggle switch with two stable states: raised (grid off) or depressed (grid on).

This would be similar to the toggle buttons for the snapping functions, shown here next to the <kbd>Toggle Grid</kbd> button:
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36888 (Git) Snap 991
Build type: Release
Branch: main
Hash: d611611055695f75a39527989224d3fb1cc51a48
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * ExplodedAssembly
  * BillOfMaterials 0.0.7.2
  * Ondsel-Lens 2024.3.26.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-04:
@furgo16  this isn't exactly what you asked for but you can hack the colour changes to be _only_ checked/unchecked, I personally like to be able to see all three states, On (green), Off but available (red) and unavailable (transparent).

[image]
[image]

With it all being Python you don't need to compile to test, see https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Draft_Grid_Observer for changes (ignore the CMakeLists.txt change). My personal stylesheet picks up the checked/unchecked for the menu icon to automatically update:

```
QMenu::icon:checked { /* appearance of a 'checked' icon */
    background: #469143;
    border: 2px #469143;
    position: absolute;
    border-radius: 2px;
}

QMenu::icon:unchecked { /* appearance of a 'unchecked' icon */
    background: #914343;
    border: 2px #914343;
    position: absolute;
    border-radius: 2px;
}
```

furgo16, 2024-05-04:
@Syres916 oh, wow, thanks! Except for the colors, that was _exactly_ what I was looking for. A few notes/questions after testing your branch, and using the Classic theme:

1. The tooltips on the button get the same background color as the button, which is probably not intended. That is, when the button is red, the tooltip is red as well. When the button is green, the tooltip is green too.
2. I think the color choice, whichever the color, is very personal, so I'd go with defaults first. To me, having a red icon on the toolbar is having something that constantly seeks my attention, like a warning light on a car's dashboard :-)
3. I've not tried to change the colors with a custom stylesheet, as I'm not familiar with how they work with FreeCAD. Therefore I cannot comment much on that part, other than expressing the opinion that probably the default in the code should be to set the Classic theme colors for checked/unchecked toggle button, and let the stylesheet override that if the user wants to. For my personal 
4. In which situation would the "unavailable (transparent)" be triggered?

Let me know if there is anything I can do to help if you're planning to submit a PR.

Syres916, 2024-05-06:
@furgo16 I've updated the branch to be just enabled/disabled and checked/unchecked so any colours only come from the user's stylesheet which deals with 1, 2 and 3.

With regard to question 4, the command should not be available when the mdi area is not the editable 3D view, in other words no open project, Techdraw sheet in view etc should disable the Toggle_Grid command.

To be honest I nolonger try to get involved in PR's with regard to subjective matters the agro isn't worth it for me. You're welcome to cherry pick any of my code and submit your own solution, I think there will be some `return` lines need adding to my code to satisfy the maintainers and stop some Lint warnings/errors being raised.

furgo16, 2024-05-07:
Thanks! I'll look into it in the next few days. Happy to submit the PR from that code, if that helps getting it merged for everyone.

closed 2024-06-13 by Roy-043