FreeCAD/FreeCAD #13530: Working plane changes only if the grid is visible

furgo16, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changing the working plane, by either selecting a face of a component, or not selecting anything, and then clicking on the working plane setup button only works if the grid is visible.

Following the steps above with the grid's visibility set to off, the working plane remains unchanged, also regardless of the visibility settings in the preferences. To make it work, the Toggle Grid button must be used until the grid is visible, and then, follow the steps above.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36888 (Git) Snap 991
Build type: Release
Branch: main
Hash: d611611055695f75a39527989224d3fb1cc51a48
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * ExplodedAssembly
  * BillOfMaterials 0.0.7.2
  * Ondsel-Lens 2024.3.26.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-04-19:
@maxwxyz thanks for triaging. Would this not rather be a https://github.com/FreeCAD/FreeCAD/labels/Bug than a https://github.com/FreeCAD/FreeCAD/labels/Feature? Especially given the described different behavior in FreeCAD 0.21.

maxwxyz, 2024-04-19:
I guess it's currently working as designed, seeing the result of the repositioning as feedback.

furgo16, 2024-04-19:
I see it differently, but it's fine to leave the label as it is.

In my view, if the grid is not visible, there is no feedback of the outcome. That is, if the grid is not visible and I change the working plane, the change did not work and I have no visual or textual feedback that it didn't.

Only when I realize that my commands are not working as expected and I turn the grid on to double check, then I have feedback of the failure.

Roy-043, 2024-04-19:
I cannot reproduce. Draft_SelectPlane does not depend on the visibility of the grid here. After using the command you can hover the button in the tray. The tooltip then shows the parameters of the plane.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

maxwxyz, 2024-04-19:
@Roy-043 I can reproduce on Windows 11 latest weekly, no mods. How do you set the plane? When using one click on the button it seems not to update. When hovering the button, the position is displayed and seems to be updated but when making the plane visible it is on the old location again.

Roy-043, 2024-04-19:
OK I can confirm now. Both the "Always show the grid" and "Show the grid during commands" preference must not be selected for the issue to occur.

Roy-043, 2024-04-19:
Changing the preferences does not affect existing views. You would have to create a new view (or open a new doc) after each pref change.

closed 2024-04-19 by Roy-043

Roy-043, 2024-04-20:
The tooltips of the preferences already explain this.

furgo16, 2024-04-21:
This now works as expected (thanks!), with a cosmetic glitch I cannot reproduce reliably: sometimes when setting the working plane (with a non-visible grid and settings "Always show the grid", "Show the grid during commands" unchecked), the grid will become visible and needs to be toggled non-visible. It does not always happen, and I've not yet pinpointed which combination of actions make it happen.

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36913 (Git) Snap 994
Build type: Release
Branch: main
Hash: 700868f8346478f88e955672e5e1bad47ae061da
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * ExplodedAssembly
  * BillOfMaterials 0.0.7.2
  * Ondsel-Lens 2024.3.26.01
  * Assembly4 0.50.12
```

PR #13540 (2024-04-19): Draft: Draft_ToggleGrid should update the grid
Fixes #13530.
commit a043cfa3 (2024-04-19): Draft: Draft_ToggleGrid should update the grid

Fixes #13530.