FreeCAD/FreeCAD #13532: Translation Error： "Preferences -> Import-Export -> IFC import " translation into Chinese error

royccg, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * Assembly4 0.50.12 (Disabled)
  * BIM 2021.12.0
  * BIM.backup1712900130.9645526 2021.12.0 (Disabled)
  * CADExchanger
  * Civil_BIM_development
  * Curves 0.6.31
  * dodo 1.0.1
  * fasteners 0.5.18
  * Glass
  * IconThemes
  * OpenDark 2024.3.13
  * parts_library
  * parts_library.backup1713183823.8482962
  * Reinforcement
  * Reporting
  * WebTools 1.0.0
  * WebTools.backup1712900673.1973262 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-04-19:
@royccg you might want to contact the Simplified Chinese translators or alternatively you should be able to fix the translation directly in CrowdIn => https://wiki.freecad.org/Localisation

royccg, 2024-04-19:
Thanks so much for reminding me of this way to fix that.

maxwxyz, 2024-04-19:
please open the issue here: https://github.com/FreeCAD/FreeCAD-translations