FreeCAD/FreeCAD #13533: QSvgWidget is not a child of QtSvg in PySide6

dov, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The source used QtSvg.QsvgWidget, but this class does not exist in PySide6.

The code needs to be ported to use QtSvgWidgets.QSvgWidget instead. See: https://doc.qt.io/qtforpython-6/PySide6/QtSvgWidgets/QSvgWidget.html#PySide6.QtSvgWidgets.QSvgWidget

### Full version info

```shell
OS: Fedora Linux 39 (Workstation Edition) (fvwm)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36442 (Git)
Build type: Unknown
Branch: main
Hash: bcf54d3153fe4cad841ff7c0b692506eaa802103
Python 3.12.2, Qt 6.6.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.8.1.dev
Locale: C/Default (C)
Installed mods: 
  * Curves 0.6.21
  * MyWorkbench
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-19:
related or duplicate to?
- https://github.com/FreeCAD/FreeCAD/issues/13303
- https://github.com/FreeCAD/FreeCAD/issues/13427

if so, please comment there

dov, 2024-04-19:
@maxwxyz Wouldn't it make sense to separate between Qt6 compilation errors, and PySide6 run time errors? Also, do you really want to group all issues together? E.g. this issue is a specific measurable problem. If you group them alltogether, how do you measure whether it is done or not?

maxwxyz, 2024-04-19:
It's ok for me as is, I'm no expert in these issues

wwmayer, 2024-04-20:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/13554

commit c4c45331 (2024-04-22): PySide6: Fixes #13533: QSvgWidget is not a child of QtSvg in PySide6 (#13554)

* Create a compatibility module QtSvgWidgets.py to handle PySide2 & PySide6
* In the Arch and Material modules use the QtSvgWidgets module
* Since Qt6 the method QFont.setWeight() doesn't accept an int any more but requires an enum. Since the call of QFont.setBold(True) sets
  a weight of 75 the extra calls of QFont.setWeight(75) can be safely removed
closed 2024-04-22 by adrianinsaval