FreeCAD/FreeCAD #13534: Spreadsheet unit calculation incorrect

xorxx, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to calculate the mass of a part but unit calculation mechanism in spreadsheet creates wrong results in numbers an units.

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * freecad.gears 1.1.0
  * Help 1.0.3
  * ToOptix
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Freecad project demonstrating wrong results in spreadsheet: [FC-Spreadsheet-Unit-Bug.zip](https://github.com/FreeCAD/FreeCAD/files/15036756/FC-Spreadsheet-Unit-Bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-19:
can reproduce but when entering the values manually, the calculation is correct.
It seems that the value from `=Sphere.Shape.Volume` gets a wrong unit.
[image]

xorxx, 2024-04-19:
It seems that the `.shape.volume` property has no unit in general. I tried it with other basic shapes like boxes and they all come without unit.

pitch-circle, 2024-04-19:
> It seems that the '.shape.volume' property has no unit in general.

That is correct. And thus it gains a default unit of mm^3. That is where your calculation is going wrong - trying to use cm^3 where it should be mm^3.

If you add the cm^3 unit in the spreadsheet to a unitless value, you have to do the conversion so the correct numerical value in cm^3 is obtained.

See: https://forum.freecad.org/viewtopic.php?p=87871#p87871 (note date of 2015).

It is also worth reading and understanding this:

https://wiki.freecad.org/Spreadsheet_Workbench#Units

maxwxyz, 2024-04-19:
So should this issue be closed or should it be rewritten to get the right unit from shape properties?

xorxx, 2024-04-22:
To me it sounds better to provide shape properties with correct units to avoid such mistakes. But I have no idea what the means for the implementation.

kaiwas, 2024-04-22:
Automatic "transformation" of dimensions can cause a lot of inconvenience.
In my opinion, it should be possible to set “default dimensions” (for example, get values only in meters) and only if another value is forced in the call field, it should be converted (for example mm ^3)

xorxx, 2024-04-23:
I would consider _"default dimensions"_ as a bug. The information _"value has no dimension"_ or even better _"value is missing dimension"_ allows to see errors in calculation.

I am not afraid of automatic conversion (the are currently working great) because they can not make a calculation result wrong. Worst consequence would be to see results in an unfamiliar unit, what is easy to heal by setting the "display dimension" forcing the conversion to whatever you want.

kaiwas, 2024-04-24:
> I am not afraid of automatic conversion

In general, I agree with you. But it all depends on the type of work.
Regarding the type of work I do, automatic conversions can be inconvenient.
For example, most often I work in millimeters and I don’t need other dimensions. I can resize from 7850mm. to 10560 or some other. And when I step over 9999+1 mm. I will have meters instead of millimeters (I didn't ask, but it's happening).
You have to constantly monitor this and manually make adjustments (indicate the dimension). For me it's a waste of time.

I described the problem here

https://github.com/FreeCAD/FreeCAD/issues/13040

xorxx, 2024-04-24:
When you add 1 mm to a length that is 9999 mm the result is 10000 mm. The only thing that FreeCad changes in this situation is the display unit, because it tries to avoid lengthy numbers.  This is not a wrong calculation result but an unwanted change of display units. So what you need is not a _default unit_ for properties but a _default display unit_ and/or _default input unit._ This is something completely different. I agree that this is a useful thing. You would need to specify default display units for all physical properties like length, area, volume, electric properties, forces etc that are used in freecad. So it will be a long table, but that should be ok.

But this is unrelated from my report here: 

1. The fact that in case of the `shape` properties NO unit is delivered, I still assume to be a bug leading to misunderstandings and wrong results.
2. Not checking the congruence of display unit settings in a spreadsheet cell to the underlying value's unit is also a bug. You can force _mm_ to be displayed a _Volts_. This is definitely wrong.

xorxx, 2024-04-24:
@maxwxyz :
> So should this issue be closed or should it be rewritten to get the right unit from shape properties?

As I elaborated [https://github.com/FreeCAD/FreeCAD/issues/13534#issuecomment-2074268959](above) this should be rewritten.

maxwxyz, 2024-04-24:
I'm in favor of rewriting.

dptsolutions, 2024-06-23:
Hi everyone, got here thanks to @maxwxyz. 

So it looks like my issue might be similar, and I read the docs about units shared in this [comment](https://github.com/FreeCAD/FreeCAD/issues/13534#issuecomment-2066997677).

But this is what I don't understand. I checked my spreadsheet and looked at the cell props of my source cell, which I had typed in 2.8mm and visually was formatted like it set the unit for the cell to millimeters. But in cell properties for that cell, the unit string field is empty:
[image]

So I am confused now. If proper usage of Spreadsheet requires me to manually set this unit string field, but it seems to recognize that when I type in _2.8mm_ it formats the cell such that the sheet knows I'm using millimeters there. So shouldn't this unit string field be set by default when I enter in 2.8mm into the cell?

xorxx, 2024-06-24:
The misunderstanding is, that the cell property is only a "DISPLAY unit". It does not change or show the unit of the cell value. The underlying unit comes from your input as in your example, or from a property of a geometry linked to this cell. With display unit you can e.g. enforce displaying a length in mm, even if it is Kilometers or you may enforce the display of a power in Joule per second instead of Watts which is physically the same.

luzpaz, 2026-01-01:
Elevating this to High priority because this is a workflow distruptor.