FreeCAD/FreeCAD #13537: Placement and Movement of Annotation Labels

maxwxyz, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Discovered in https://github.com/FreeCAD/FreeCAD/pull/9750#issuecomment-2066481912

An annotation cannot be moved when viewing the model from one of the base views (front, top, right,...). All other angles work fine.
When moving the annotation label, it is moved at an increasing distance to the model, it is not possible to manually move it closer to the model

[image]

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/40ca4810-f737-4d9d-8936-e28d2d1737c6

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * CfdOF.backup1712690352.9917417 1.25.10 (Disabled)
  * CurvedShapes 1.0.9
  * Curves 0.6.33
  * Curves.backup1713373721.4125743 0.6.30 (Disabled)
  * Defeaturing 1.2.2
  * fasteners 0.5.17
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad.gears.backup1712859125.6461835 1.2.0 (Disabled)
  * freecad_metal_workbench 0.0.1
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.15
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Core

### Anything else?

Related:
- https://github.com/FreeCAD/FreeCAD/issues/13116

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-15:
I can't reproduce the second part of this issue (might be fixed already) but I can reproduce the first one.

Can you add "Measurement" to the tile for clarity?

@hlorus FYI

maxwxyz, 2024-06-16:
@FEA-eng I've updated the issue. It's not an issue from measurement but any annotation.

maxwxyz, 2024-07-26:
@FEA-eng the second part was the placement of the measurement label too far away from the model. 
It is still placed way off. It seems to be dependent of the model scale. This is the BIM example file:

https://github.com/user-attachments/assets/df8bd6fb-ae48-43b0-adfb-3a98e0c64fc1

@hlorus FYI

luzpaz, 2025-05-17:
@hlorus still want this ticket assigned to you ?

hlorus, 2025-05-18:
I'm afraid i won't currently find the time to look into this.

luzpaz, 2025-11-25:
Hey @3x380V, does wmayer have a fix for this in his branch ?