FreeCAD/FreeCAD #13539: TechDraw: "No default template found" error in weekly

FEA-eng, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm unable to use TechDraw in the newest Windows weekly. When I click the _Insert Default Page_ button, I get the "No default template found" error and the following warning message appears in the report view:

`15:16:36  Template File: E:\Program Files\FreeCAD_weekly-builds-36874-2024-04-17-conda-Windows-x86_64-py311\data\Mod/TechDraw/Templates/A4_LandscapeTD.svg is not readable`

@WandererFan FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dplanella, 2024-04-19:
I'm not sure if this is related, but on https://github.com/FreeCAD/FreeCAD/pull/13368 I renamed the custom `TD` and `m52` TechDraw templates to include an underscore for consistency. I could imagine that if you had your default template set to any of those, you'd run into this error.

WandererFan, 2024-04-20:
I struggle with windows issues, but is the presence of both forward and backward slashes in the file path relevant?

What is in your preference for the default template?

FEA-eng, 2024-04-20:
The paths are empty:

[image]

It was working normally in the previous weekly builds.

maxwxyz, 2024-04-30:
can confirm, same behavior.
Discussion: https://forum.freecad.org/viewtopic.php?p=756351
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13
```

maxwxyz, 2024-05-01:
[image]

prokoudine, 2024-05-01:
It's the same on Linux (Ubuntu 23.10).

WandererFan, 2024-05-18:
fixed by https://github.com/FreeCAD/FreeCAD/pull/13784

closed 2024-05-18 by WandererFan

FEA-eng, 2024-05-18:
> fixed by #13784

Yes, but the default template is a blank page now so it’s kind of a regression. Do you plan to fix that? Should I create a new issue?