FreeCAD/FreeCAD #13542: Sketcher: symmetry doesn't work properly for arc midpoints

FEA-eng, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you try to constrain an endpoint of a line to the middle point of an arc (point in the middle of its length) using the Symmetry constraint then strange stuff will happen. Either the arc will become tiny with no way to increase its size or the line will be forced to stay perpendicular to it. It may not work or fail to converge as well. In any case, there's no easy way to just do this in FreeCAD:

[image]

without having to use a tedious workaround with construction lines:

[image]

If the symmetry constraint can't be extended to work in such cases then there are other ways like adding midpoints to arcs when they are created. Ideally, it should be also possible to use auto constraint here. Snapping to the arc midpoint already works.

Old forum discussion: https://forum.freecad.org/viewtopic.php?t=50740

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-19:
I don't really understand the issue, could you add a demo file or another way how I can reproduce it? Of course it won't work when selecting the end point of the line and both arc endpoints and applying a symmetry constraint, as this would result in a point symmetry, where I understand you need a symmetry line through the arc center and the arc midpoint.

FEA-eng, 2024-04-19:
@maxwxyz What I mean in this issue is that I’ve tried various different ways and the symmetry constraint can’t be used like in the SolidWorks example so it’s a current limitation (although not suggested by the documentation or possible selections). My goal is just to have an arbitrarily oriented line attached to the midpoint of an arc.

maxwxyz, 2024-04-19:
Ok it seems not suitable for a symmetry constraint in my opinion.
@PaddleStroke this looks like it is fixed with your recent midpoint constraint?

FEA-eng, 2024-04-19:
@maxwxyz The problem is that the new midpoint auto constraint applies the symmetry constraint so it’s not suitable for this. The easiest way could be to just add another point (maybe appearing only when snapping) in the middle of an arc.

maxwxyz, 2024-06-13:
Is this still relevant or fixed?

PaddleStroke, 2024-06-13:
I think still relevant.
Currently middle of arcs only have a snap. But no auto-constraint is created. Symmetric is not suited.

This would need to add a new type of constraint I think.

Adding a point to arcs is difficult because arcs already have start/end/mid points, so it would need to add one more type of points, ie changing a lot of things including the element widget. A real PITA. Also it is less good as a solution than adding a constraint.

Maybe it is on your roadmap @FlachyJoe ?