FreeCAD/FreeCAD #13544: TechDraw: arc length dimension doesn't follow the standards

FEA-eng, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be good to improve the arc length dimension so that it follows the standards. Currently, it looks like this:

[image]

while it should look like this:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-19:
> Could you please provide a reference for the figure you include (i.e. from which ISO Standard does it come)? Does it say in the Standard which character should be used for the 'hat' over the numerical value?

It's from a Polish book, I don't have access to the standards themselves. But this is also how other CAD software dimensions arc length. For example, in SolidWorks:

[image]

Another example from the book:

[image]

maxwxyz, 2024-04-19:
same icon as proposed is implemented in the arc length dimension constraint in the Sketcher WB

PaddleStroke, 2024-04-19:
Yes the smart tool can create this dimension. Changing its rendering shouldn't affect my PR

pierreporte, 2024-04-19:
The most used reference book in France (*Guide du dessinateur industriel*) mentions the current style as the only one. It doesn’t give the standard, as it sometimes do, and it doesn’t mention the “correct” way, which is strange.

[image]

The other book (*Guide des sciences et technologies industrielles*) only show the same dimension style as well but it gives two standards: NF EN ISO 2538-1 and NF EN ISO 2538-2, which are basically **ISO 2538-1/2**.

pierreporte, 2024-04-19:
I found the standard : **ISO 129**.

WandererFan, 2024-04-20:
From ISO 129-1:2004(E) page 18:
[image]

WandererFan, 2024-04-20:
Without digging into the code, it looks like we are using Intersection: “∩” (U+2229), but should be using Arc: “⌒” (U+2312).

maxwxyz, 2024-04-20:
@WandererFan in Sketcher WB it's also used `std::string("◠ ")`
from:
- https://github.com/FreeCAD/FreeCAD/pull/12602

maxwxyz, 2024-04-24:
The current standard ISO 129-1:2022 defines the arc length symbol in relation to the text height:
[image]

FEA-eng, 2024-04-24:
Then I wonder why SolidWorks uses a different style. It would be good to check other software. Maybe they are following other standards (like ASME?).

pierreporte, 2024-04-24:
In any case users should be able to make drawings following ISO or ASME. This should be a drawing settings.

pitch-circle, 2024-04-29:
> Maybe they are following other standards (like ASME?).

They are. The ASME Standard is ASME Y14.5-2009 (this is mentioned in the TechDraw documentation). Paragraph 1.8.3, Figure 1-28.

Y14.5 Truetype font is here: https://cstools.asme.org/csconnect/CommitteePages.cfm?Committee=C64041000&Action=40068

The 'hat' for an arc dimension is U+00AE (Alt+0174 in Windows).

It might be better to refer to both the above standard (can be found as a pdf online) and to ISO129 (also available online) in advance of raising further bug reports. Your Polish textbook is a wonderful work (I bought a copy) but the two standards are far better reference works when it comes to FC development.

A more useful way to write the current issue (one that puts all the relevant information at the fingertips of the developers should they choose to take up the issue) is "TechDraw correctly shows arc dimensions in ISO 129 format but does not yet appear to support ASME Y14.5 conventions for the same dimensions (see ASME Y14.5 para 1.8.3 and fig 1.28). Note that the 'hat' character for these dimensions can be found at the following URL..."

maxwxyz, 2024-04-29:
Ideally there should be one preference to set the standard which should be used for all TechDraw objects: dimensions, line types/styles, projection method, section cuts, hatching, notation, page template/sizes, display of hidden lines...
Like the units in FreeCAD: either work in metric, imperial or whatever there is.