FreeCAD/FreeCAD #13547: Addon Manager missing from Tools menu

MaggieLeber, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Believe I've searched for an existing problem correctly, but I'm a total noob with FreeCAD.

Just installed 0.21.2 on my Windows 10 machine... AddonManager isn't present on the Tools menu. Attempting to use Help 
launches this dialog: 
[image]

Pushing the "Open Addon Manager" button displays:

```18:50:26  During initialization the error "No module named 'queue'" occurred in D:\FreeCAD 0.17\Mod\AddonManager\InitGui.py
18:50:26  Please look into the log file for further information
18:50:27  During initialization the error "module 'FreeCAD' has no attribute 'EndingAdd'" occurred in D:\FreeCAD 0.17\Mod\Image\InitGui.py
18:50:27  Please look into the log file for further information
18:50:28  During initialization the error "'PySide2.QtCore.qRegisterResourceData' called with wrong argument types:
  PySide2.QtCore.qRegisterResourceData(int, str, str, str)
Supported signatures:
  PySide2.QtCore.qRegisterResourceData(int, bytes, bytes, bytes)" occurred in D:\FreeCAD 0.17\Mod\Plot\InitGui.py
18:50:28  Please look into the log file for further information
18:50:28  During initialization the error "'PySide2.QtCore.qRegisterResourceData' called with wrong argument types:
  PySide2.QtCore.qRegisterResourceData(int, str, str, str)
Supported signatures:
  PySide2.QtCore.qRegisterResourceData(int, bytes, bytes, bytes)" occurred in D:\FreeCAD 0.17\Mod\Ship\InitGui.py
18:50:28  Please look into the log file for further information
18:50:31  Unknown command 'Std_AddonMgr'```

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-20:
I cannot reproduce. Is it available in the recent weekly builds for you: 
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2024-04-20:
> 18:50:26  During initialization the error "No module named 'queue'" occurred in D:\FreeCAD 0.17\Mod\AddonManager\InitGui.py
> 18:50:26  Please look into the log file for further information
> 18:50:27 [...]

0.17? Could you provide your _PATH_ environment variable and launch _FreeCAD_ with `-l `option [^1] and upload the log file?

[^1]: https://wiki.freecad.org/Debugging

MaggieLeber, 2024-04-20:
The directory name contains the original 0.17 because the installer inserts it and I wanted to get current; I'd intended to rename it later. 

System path:
```D:\Python312\Scripts\;D:\Python312\;D:\Java\TemurinJDK20\bin;D:\Java\TemurinJDK17\bin;D:\ImageMagick-6.8.8\ImageMagick-7.1.0-Q16-HDRI;C:\Program Files\Eclipse Foundation\jdk-11.0.12.7-hotspot\bin;D:\Java\OpenJDK8U-jre_x64_windows_hotspot_8u202b08\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\jEdit;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;D:\apache-maven-3.6.0;D:\Go\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PowerShell\7-preview\preview;D:\PuTTY\;d:\JetBrains\IntelliJ IDEA 2018.3.2\bin;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;%SYSTEMROOT%\System32\OpenSSH\;D:\sbt\bin;C:\Program Files (x86)\Calibre2\;D:\scala\bin;d:\Git\cmd;D:\nodejs\;C:\ProgramData\chocolatey\bin;D:\Windows Kits\10\Windows Performance Toolkit\;D:\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;D:\Pandoc\;D:\scala-cli\;```

User path
```D:\Rustup\Cargo\bin;C:\Users\maggie\.cargo\bin;d:\Ruby26-x64\bin;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;D:\Microsoft VS Code\bin;%USERPROFILE%\go\bin;D:\JetBrains\IntelliJ IDEA 2018.3.2\plugins\maven\lib\maven3\bin;C:\Users\maggie\AppData\Roaming\npm;d:\Midnight Commander;d:\protodot;```

Ah, the log file was a merry chase. Here we go: 

```Msg: FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Time = 2024-Apr-20 13:45:06
Log: AboutImage = freecadabout
Log: AppDataSkipVendor = true
Log: AppHomePath = D:/FreeCAD 0.17/
Log: AppIcon = freecad
Log: AppTempPath = C:\Users\maggie\AppData\Local\Temp\\
Log: BOOST_VERSION = 1_67
Log: BinPath = D:/FreeCAD 0.17/bin\
Log: BuildRepositoryURL = git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-21
Log: BuildRevision = 33771 (Git)
Log: BuildRevisionBranch = releases/FreeCAD-0-21
Log: BuildRevisionDate = 2023/11/14 02:33:39
Log: BuildRevisionHash = b9bfa5c5507506e4515816414cd27f4851d00489
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 21
Log: BuildVersionPoint = 2
Log: Console = 0
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Debug = 0
Log: DocPath = D:/FreeCAD 0.17/doc\
Log: EIGEN_VERSION = 3.3.90
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.21.2
Log: LoggingConsole = 1
Log: LoggingFile = 1
Log: LoggingFileName = C:\Users\maggie\AppData\Roaming\FreeCAD\FreeCAD.log
Log: MaintainerUrl = http://www.freecad.org/wiki/Main_Page
Log: OCC_VERSION = 7.6.3
Log: PATH = D:\Python312\Scripts\;D:\Python312\;D:\Java\TemurinJDK20\bin;D:\Java\TemurinJDK17\bin;D:\ImageMagick-6.8.8\ImageMagick-7.1.0-Q16-HDRI;C:\Program Files\Eclipse Foundation\jdk-11.0.12.7-hotspot\bin;D:\Java\OpenJDK8U-jre_x64_windows_hotspot_8u202b08\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\jEdit;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;D:\apache-maven-3.6.0;D:\Go\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PowerShell\7-preview\preview;D:\PuTTY\;d:\JetBrains\IntelliJ IDEA 2018.3.2\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;D:\sbt\bin;C:\Program Files (x86)\Calibre2\;D:\scala\bin;d:\Git\cmd;D:\nodejs\;C:\ProgramData\chocolatey\bin;D:\Windows Kits\10\Windows Performance Toolkit\;D:\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;D:\Pandoc\;D:\scala-cli\;D:\Rustup\Cargo\bin;C:\Users\maggie\.cargo\bin;d:\Ruby26-x64\bin;C:\Users\maggie\AppData\Local\Microsoft\WindowsApps;D:\Microsoft VS Code\bin;C:\Users\maggie\go\bin;D:\JetBrains\IntelliJ IDEA 2018.3.2\plugins\maven\lib\maven3\bin;C:\Users\maggie\AppData\Roaming\npm;d:\Midnight Commander;d:\protodot;
Log: PYSIDE_VERSION = 5.15.0
Log: PYTHON_VERSION = 3.8.10
Log: PythonSearchPath = D:\FreeCAD 0.17\bin\python38.zip;D:\FreeCAD 0.17\bin\DLLs;D:\FreeCAD 0.17\bin\lib;D:\FreeCAD 0.17\bin
[... cut ...]
Log: Toolbars restored
Log: 3Dconnexion device not attached.
Log: Init: Entering event loop
Log: Init: Executing event loop...
Log: Init: Processing command line files
Log: Hide main window
Log: Show main window
Log: Hide main window
Log: Show main window
Log: Hide main window
Log: Finish: Event loop left
Log: Destruct Gui::Application
Log: FreeCAD terminating...
Log: Saving system parameter...
Log: Saving system parameter...done
Log: Saving user parameter...
Log: Saving user parameter...done
```

xtemp09, 2024-04-20:
The old installation messed up the new one. @MaggieLeber, delete _FreeCAD_, delete the configuration files[^1][^2], install _FreeCAD_ anew into a normal folder.

----

I'd also delete all these things in _PATH_.

One can launch a program in a “clean” environment or with a special _PATH_. The simple way to do this is to create a bat (or cmd) file and write there something like the following:

```ruby
set PATH="C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;"
FreeCAD.exe
```

[^1]: c:\Users\maggie\AppData\Local\FreeCAD
[^2]: c:\Users\maggie\AppData\Roaming\FreeCAD

MaggieLeber, 2024-04-20:
OK, we'll wipe it all and start over. 

"All those things in path" are there for reasons, though.
 
Window programs in general do a crappy job of dealing with Users\home that aren't on the boot drive, the boot drive accumulates cruft that shouldn't be there. Pain point for SSD boot with spinny mass storage.  

OK, there were two uninstallers in the Windows apps settings, ran them both and one of them even removed the AppData stuff.
Fresh install running fine now, including Addon Manager. 

Thanks for your help, I dropped some bucks in the donation box to show my appreciation.

closed 2024-04-20 by MaggieLeber

xtemp09, 2024-04-21:
> Window programs in general do a crappy job of dealing with Users\home that aren't on the boot drive

If I were you, I would have used symbolic links or junctions, leaving standard paths intact.