FreeCAD/FreeCAD #13549: Freecad current 3rParty salomesmesh don't compile with boost-1.85 on FreeCAD current

Tarot67, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

freecad/src/3rdParty/salomesmesh/src/SMESH/DriverGMF.cpp:58:43: error: ‘extension’ is not a member of ‘boost::filesystem’
   58 |     std::string ext  = boost::filesystem::extension(fileName);
      |                                           ^~~~~~~~~
make[2]: *** [src/3rdParty/salomesmesh/CMakeFiles/SMESH.dir/build.make:96: src/3rdParty/salomesmesh/CMakeFiles/SMESH.dir/src/SMESH/DriverGMF.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3611: src/3rdParty/salomesmesh/CMakeFiles/SMESH.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
bash-5.2$ elvis src/3rdParty/salomesmesh/CMakeFiles/SMESH.dir/flags.make 
reading src/3rdParty/salomesmesh/CMakeFiles/SMESH.dir/flags.make

### Full version info

```shell
I m in branch main (dev)
last commit is:
eb59682a536e3228a81ad3dc2d5215ee58a56c1d
```

### Subproject(s) affected?

None

### Anything else?

https://www.boost.org/doc/libs/1_84_0/libs/filesystem/doc/v4.html i see that. There break change in boost library.
But I'm not sure that the problem.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-20:
Does this work:
`std::string ext = boost::filesystem::path(fileName).extension().string();`
?

Tarot67, 2024-04-20:
Yes it's working!

Tarot67, 2024-04-20:
I use this expression:
std::string ext = boost::filesystem::path(fileName).extension().string();

compile now. I think you must use pragma ;.( . 

Thank you very much

wwmayer, 2024-04-20:
Pragma? I guess you meant an `#ifdef / #else / #endif`  but this is not needed. The function of the convenience.hpp file did just what I have posted.

Tarot67, 2024-04-20:
Yes pragma it is exactly that you describe. But if no need it is better!. Thank you again.

commit 9c84f3a7 (2024-04-22): Merge pull request #13570 from wwmayer/issue_13549

Fixes #13549: smesh doesn't compile with boost 1.85
commit f555a25f (2024-04-22): Fixes #13549: smesh doesn't compile with boost 1.85
closed 2024-04-22 by sliptonic