FreeCAD/FreeCAD #13552: "Move to another body" is unnecessarily restrictive

jwueller, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, it just shows this message and aborts:

> Some of the selected features have dependencies in the source body

It's really hard to figure out which dependency is the problem in a complex part. In essence, moving to a different body should be equivalent to a cut/paste operation. Based on this, here is my proposal:

1. Allow the selected items could be moved anyway, so that problems could at least be resolved manually afterwards.
2. Utilize the cut/copy dialog, where you can select which of the dependencies to move with the selection.

I think 1. is an easy win, because it's likely trivial to implement and at least makes the problem solvable for the user. 2. might require more work, but would be nice to have.

If none of these are desired, the dialog should _at least_ show which dependency caused the conflict, so it can be addressed. But I think the dialog could just go away entirely by just allowing the move, and dealing with problems as they come up, like you would in most other cases.

I'll attempt a PR for this.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36902 (Git)
Build type: None
Branch: main
Hash: b331a9f7a229581a7592f02a1241eaa572126e3c
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-20:
is this related to: 
- https://github.com/FreeCAD/FreeCAD/issues/9725

jwueller, 2024-04-20:
It's related, but it's not the same problem. Dragging it over just does nothing, while explicitly choosing the context menu option produces a warning and aborts. It seems they use different implementations, but I might get a chance to look at both when I fiddle with the implementation a bit more.

maxwxyz, 2024-04-20:
great, should I assign you to this issue?

jwueller, 2024-04-20:
Sure, I'll at least take a look.