FreeCAD/FreeCAD #13555: Sketcher: OVP line with 2 modes not displayed

maxwxyz, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke FYI
When using OVP with the setting `dimension only` the OVP of the line tool with mode 2 points are not displayed. 
When using the setting for all OVP the OVPs are also displayed for the second point.
[image]

Also, when the first point is placed a `M`for cycling the mode resets the tool and the first point has to be placed again. I get the behavior for `all OVP` but `dimension only` could keep the first point. I guess this part is a FR as all tools handle this the same way.

### Full version info

```shell
OS: macOS 13.6.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.2.0
  * OpenDark 2024.3.13
  * CfdOF 1.25.11
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.17
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.11
  * Curves 0.6.34
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-04-22:
2 points mode is using only positional OVP. So indeed if you are using the option 'dimensional only', you will see not OVP in the 2 points mode.
Do you think this mode should show its OVP when 'only dimensional' even if they are positional ?

maxwxyz, 2024-04-22:
mh, never thought of it this way.
Maybe dimensional could be renamed to `not the starting point` as this was the thing I though it did 😄

FEA-eng, 2024-06-15:
It seems to be working as expected (no positioning OVP when the setting is Dimensional only). So maybe the issue could be closed?

maxwxyz, 2024-06-16:
I'm closing this in favor of https://github.com/FreeCAD/FreeCAD/issues/14658
For me personally, dimensional is also a horizontal distance of a point so I thought dimensional is only active after the first start point.

closed 2024-06-16 by maxwxyz