FreeCAD/FreeCAD #13557: Sketcher: Dimensions not visibly when viewing sketch from behind

maxwxyz, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When rotating the sketch the dimensions disappear as soon as the sketch plane is mostly viewed from the back side. From behind the dimension is not visible and you cannot select or change it.
It does not depend if `view section` is active.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/50252f41-6370-4e72-9668-a7304869b383

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.34
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenDark 2024.3.13
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Sketcher

### Anything else?

Issues related to dimension visibility:
- https://github.com/FreeCAD/FreeCAD/issues/13318
- https://github.com/FreeCAD/FreeCAD/issues/10100

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-20:
@PaddleStroke FYI

PaddleStroke, 2024-04-22:
I cannot reproduce. Perhaps its specific to the file? Can you try with an empty file @maxwxyz ?

maxwxyz, 2024-04-22:
Yes, same problem with an empty file in the current weekly:
```OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.11

```

maxwxyz, 2024-05-17:
Still present:

https://github.com/FreeCAD/FreeCAD/assets/6246609/4f7de066-29b3-4304-819f-99fe50eab929

@FlachyJoe FYI

FlachyJoe, 2024-05-17:
#14053 fix this too

PR #14053 (2024-05-20): Gui: fix SoDatumLabel arrow and text disappear
Fix #14049 
Fix #13557
arrow drawn CCW and text was not shown in every cases.
closed 2024-05-20 by chennes