FreeCAD/FreeCAD #13561: Material: Exception is raised if a group element of the Material list is selected

wwmayer, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting a group element of the Material list e.g. Favorites, Recent, System, Appearance, ... then an exception is raised that's not handled inside the slot function where it occurs.

Because of this the exception must be handled inside GUIApplication::notify() where the following error message is printed:
```
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: FreeCAD Exception
```

So, there are two things that must be improved:
* The default constructor of MaterialNotFound should set the text "Material not found" to make it easier to locate future problems
* The slot function MaterialTreeWidget::onSelectMaterial should not call getMaterialManager().getMaterial if the passed uuid is an empty string:
```
if (!uuid.isEmpty()) {
    Q_EMIT materialSelected(getMaterialManager().getMaterial(uuid));
}
```

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36911 (Git)
Build type: debug
Branch: main
Hash: 90a9020ed4bf7f630a6a822871316d13159d55a4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2025-07-21:
And i was looking in the editor, not the MaterialTreeWidget. It appears to be fixed. The suggested changes have been incorporated.

FEA-eng, 2025-09-20:
@davesrocketshop Can it be closed?