FreeCAD/FreeCAD #13562: FEM: hardcoded colors in solver job

maxwxyz, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally posted here: https://github.com/obelisk79/OpenDark/issues/40
[image]

The text is not legible with dark themes and the color is not exposed so a theme could style it / change color.
@FEA-eng FYI

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git) AppImage
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.14
  * sheetmetal 0.3.12
  * Rocket 3.2.4
  * kicadStepUpMod 10.21.4
  * fasteners 0.4.70
  * Render 2023.8.18
  * BIM 2021.12.0
  * FreeCAD-Reinforcement-added_translations.zip
  * FreeCAD-Reinforcement-master.zip
  * Reinforcement
  * Alternate_OpenSCAD 1.0.0
  * Woodworking 0.21.30486
  * dodo 1.0.0
  * Alternate_OpenSCAD.backup1691601718.9594336.zip
  * OpenDark 2023.10.30
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lyphrowny, 2024-04-20:
The function responsible is this one: https://github.com/FreeCAD/FreeCAD/blob/700868f8346478f88e955672e5e1bad47ae061da/src/Mod/Fem/femtaskpanels/task_solver_ccxtools.py#L189-L195

How to expose colors? Any examples in other WBs?
Do the colors need to be exposed or can they be assigned by accessing the theme? For example `ThemeConfig.ErrorColorText`

___

These functions may cause the same problem:
* https://github.com/FreeCAD/FreeCAD/blob/700868f8346478f88e955672e5e1bad47ae061da/src/Mod/Fem/femtaskpanels/task_mesh_gmsh.py#L155-L161
* https://github.com/FreeCAD/FreeCAD/blob/700868f8346478f88e955672e5e1bad47ae061da/src/Mod/Fem/femsolver/reportdialog.py#L61-L72

maxwxyz, 2024-04-21:
@obelisk79 what do you need to style color / how to use text color from the preferences?

obelisk79, 2024-04-22:
Just use the same variables as the Python Editor/report view should do the trick. Plus it would make things generally more consistent across the application.

Syres916, 2024-04-25:
The CfdOF Wb already has this implemented, see https://github.com/jaheyns/CfdOF/blob/master/CfdOF/Solve/TaskPanelCfdSolverControl.py#L81 and https://github.com/jaheyns/CfdOF/blob/master/CfdOF/CfdTools.py#L370

I personally added bold to the time just to make it stand out when I'm benchmarking a model on different pieces of kit.

[image]

Syres916, 2024-04-27:
@FEA-eng would you like to try the following changes https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:FEM_TaskPanel_Colours?expand in order to give a very similar output to the CfdOF screenshot above?

FEA-eng, 2024-04-27:
@Syres916 I tested it, looks good.

PR #13684 (2024-04-29): [FEM] CCX Solver and Gmsh Mesh Task Panel Colour output improvements
Fixes #13562 , existing output window example: https://github.com/FreeCAD/FreeCAD/issues/13562#issue-2254619551

using this PR:

[image]