FreeCAD/FreeCAD #13565: TechDraw: 'Position section view' should not need a tool.

PaddleStroke, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

'Position section view' should not need a tool.

When dragging the view, there should be a snapping to correct positions when close enough. It would be more intuitive and would not require a tool.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-15:
Let me just clarify here that snapping in TechDraw is implemented (with #13659) but doesn't work for oblique section views so the TechDraw_ExtensionPositionSectionView was left and the issue is not fully resolved yet.

WandererFan, 2024-10-19:
AFAIK, oblique section snapping works now.  This can be closed.

closed 2024-10-19 by WandererFan