FreeCAD/FreeCAD #13566: TechDraw: Position chain dimensions should not need a tool.

PaddleStroke, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

Techdraw has 2 tools that let you align correct dimensions.
These tools are not necessary. If we had a snapping when dragging dimensions we could make it simple to align dimensions.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-20:
@PaddleStroke if possible, take into account:
- https://github.com/FreeCAD/FreeCAD/issues/11636

    Create dimensions with an equal (defined) spacing to the model (e.g. all 15mm from the model edge outwards)
    Snap the text on the dimension to the center of the dimension arrows (default, when moving snap text/value too)
    Align the dimension (arrows) while moving to other dimensions (snapping) (this tool you've mentioned in this your issue)
    Move the dimension without changing the local (relative) position of the dimension text (so it stays in the center)

PaddleStroke, 2024-04-20:
Ah yes it's more or less a duplicate of your issue

maxwxyz, 2024-04-20:
Yes but I've not mentioned the removal of the tool

WandererFan, 2024-04-21:
What is really needed here is a DimensionGroup so that the dimensions would continue to behave as a group after creation.  The current tool aligns the dimensions once, then if the dimension text or dimension points change, you have to do the alignment again.

closed 2024-05-30 by PaddleStroke