FreeCAD/FreeCAD #13568: Move with host property not working as expected

furgo16, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've been watching the "[BIM Getting started](https://youtu.be/Jc12_G5DFF0?si=6CrF7XAWOBNSMl2F&t=1397)" video and following the steps there to create a basic building:

The video mentions and illustrates that when grouped under e.g. a Level (Arch_Floor), the children will move with the host when that host is moved and the children's "[url=https://wiki.freecad.org/Arch_Component]Move with host[/url]" property is set to True.

However, I'm finding that that's not the case. If I:

1. Select the upper floor in the tree, with its children's "Move with host" property set to true
2. Move it either using the "Move" tool or manually setting its Placement > Position > z property to e.g. 6m

The parent floor is moved, but its children do not move with it. See attachment for the file I was using to test this. Am I missing something?

[attachment=0]Playpen - BIM Tutorial continue.FCStd[/attachment]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36888 (Git) Snap 991
Build type: Release
Branch: main
Hash: d611611055695f75a39527989224d3fb1cc51a48
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
  * Curves 0.6.29
  * ExplodedAssembly
  * BillOfMaterials 0.0.7.2
  * Ondsel-Lens 2024.3.26.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Arch

### Anything else?

1. File to reproduce: [Playpen - BIM Tutorial GH.zip](https://github.com/FreeCAD/FreeCAD/files/15050659/Playpen.-.BIM.Tutorial.GH.zip)
2. Initially [reported on the forum](https://forum.freecad.org/viewtopic.php?p=754160#p754160)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-21:
It works for me, when the children are directly under the parent object. In you case they are in a group separating them from the parent.
I have not much experience with arch @yorikvanhavre is this the expected behavior or a bug?

kaiwas, 2024-04-21:
I would not like this behavior to be intended.
A group is simply an element for convenient cataloging. It should not spoil the overall logic.
I perceive this behavior as a bug.

kaiwas, 2024-04-21:
Group "window" objects are not affected in the same way as walls. But when you place a window in a group, another additional entity is created.
In addition, I discovered one more feature. If we “move” to a floor, then nothing happens.
Video example.
Very interesting

https://github.com/FreeCAD/FreeCAD/assets/32273833/5f839bbb-8628-4668-ac5d-c819ae961b12

furgo16, 2024-04-21:
Thank you for testing. Notice that on the "[BIM Getting started](https://youtu.be/Jc12_G5DFF0?si=6CrF7XAWOBNSMl2F&t=1397)" video linked in the issue description, all components are inside groups that are in turn inside the level ([Arch_BuildingPart](https://wiki.freecad.org/Arch_BuildingPart)).

So it seems that at least at the point the video was made, the intended behavior was for "Move with host" to work as expected, regardless of whether the components were direct children of the level or indirect children inside a group that was a direct child.

furgo16, 2024-04-21:
[This issue](https://github.com/FreeCAD/FreeCAD/issues/12921) describes a similar behavior (behaves differently inside or outside groups) with another `Arch_BuildingPart` property.

PR #13589 (2024-04-30): Arch: Fixed move with host
BuildingParts did not move their children if they were inside groups
fixes #13568
commit 25c1c6c3 (2024-04-30): Arch: Fixed move with host - fixes #13568
closed 2024-04-30 by Roy-043