FreeCAD/FreeCAD #13572: TechDraw: Add Workpiece Edge Symbols (ISO 13715)

maxwxyz, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great to support annotations for workpiece edge symbols (ISO 13715)
[image]
From https://github.com/FreeCAD/FreeCAD/issues/11671

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.34
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-08-30:
@WandererFan @maxwxyz Can you supply the ISO? Perhaps I will make this... but no promises

maxwxyz, 2024-08-31:
@benj5378 I don't have a digital copy but these are the symbol specs:
[image]

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

pierreporte, 2025-09-09:
@5axes You need to open a PR if you want the reward.

rehan-ahmed-aix, 2026-02-22:
@yorikvanhavre @chennes In order to make sure the feature looks exactly as it should and does everything needed, I really need to read the actual standard. 

I was told I could ask the FPA for access to standards? I was told I could ask the FPA for access to standards? Could you help me get a copy of ISO 13715? I just want to double check the official spec so I don't miss any edge cases or specific dimensioning rules