FreeCAD/FreeCAD #13573: TechDraw: Section cuts / representation of section lines / arrows (ISO 128-40)

maxwxyz, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great to support thesSection cuts / representation of section lines / arrows without a continuous section line. (ISO 128-40) [image]

The idea was to remove the dotted/dashed line (marked with arrows) and only leave the edges, where the complex section cut makes a turn (as in the original post with a reference image from the ISO):
[image]

From: https://github.com/FreeCAD/FreeCAD/issues/11671

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.34
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-04-21:
This would involve allowing a "noline" option to the available line style choices.  Not sure off the top of my head how big a change that would be.

WandererFan, 2024-05-19:
PR  [TD]assorted Preference updates (fix #12686, #12239) #14138 adds preference for display of section cut line.

The arrow position depends on choice of section line convention parameter.  ASME calls for arrows on one side of the line, ISO calls for arrows on the other side.

closed 2024-06-10 by WandererFan